360Controller/360Controller #356: Xbox One Wired Headset

jamesah2008, 2016-09-16:
How might we get started adding the wired headset that plugs into the Xbox One controller to this driver? I know nothing of C and am a total new but point me in the right direction and maybe I can help?

FranticRain, 2016-09-16:
Figure out how to open an isochronous pipe using IOUSBHostPipe. Sniff USB packets from Windows 10 playing a standard square wave through the headphone jack. Decode the packets that the system sends to the driver over USB. Duplicate that code, but using Apple's audio input format. Also figure out how to advertise the controller as an audio device.

And another free tip, the init packet for a controller with an audio jack connected is { 0x05, 0x23 }, not { 0x05, 0x20 } like it was before.

jamesah2008, 2016-09-16:
Wow. Holy bizness. Decoding the packets sounds tough.

FranticRain, 2016-09-16:
From what I've glanced at, it doesn't seem too bad. It's not a very large packet. Nothing like getting the wireless adapter to work.