360Controller/360Controller #357: Kernel panic running OS X Sierra

christ776, 2016-09-23:
Hi, I used to have this issue with the Betas, and still happens with the public release of macOS Sierra from the AppStore. I was having 0.16.3 , removed it, installed 0.16.4 and still got the kernel panics right after unplugging the dongle.

```
Anonymous UUID:       5C49430D-212B-8D84-6799-C73E6F2F0EF3

Fri Sep 23 00:57:10 2016

*** Panic Report ***
panic(cpu 4 caller 0xffffff8010409f4a): Kernel trap at 0xffffff80108a0c37, type 13=general protection, registers:
CR0: 0x000000008001003b, CR2: 0x000000010a421000, CR3: 0x0000000013fb4000, CR4: 0x00000000001627e0
RAX: 0xc0ffee0b0607296a, RBX: 0xffffff8034de3400, RCX: 0xffffff803582ba40, RDX: 0xffffff8030a35e60
RSP: 0xffffff91fcfe3e10, RBP: 0xffffff91fcfe3e10, RSI: 0x0000000000000000, RDI: 0xffffff8034de3400
R8:  0x000000000000ffff, R9:  0x0000000000010001, R10: 0xffffff8010a66fb0, R11: 0xffffff801093bee4
R12: 0xffffff803582ba88, R13: 0xffffff803582ba40, R14: 0xffffff8030a35e60, R15: 0xffffff803582ba40
RFL: 0x0000000000010246, RIP: 0xffffff80108a0c37, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x000000010a421000, Error code: 0x0000000000000000, Fault CPU: 0x4, PL: 0

Backtrace (CPU 4), Frame : Return Address
0xffffff81eeb55df0 : 0xffffff80102f748c 
0xffffff81eeb55e70 : 0xffffff8010409f4a 
0xffffff81eeb56050 : 0xffffff80102a9743 
0xffffff81eeb56070 : 0xffffff80108a0c37 
0xffffff91fcfe3e10 : 0xffffff7f9179cb4c 
0xffffff91fcfe3e30 : 0xffffff8010899d7e 
0xffffff91fcfe3e80 : 0xffffff801089b26a 
0xffffff91fcfe3f10 : 0xffffff801089b0d6 
0xffffff91fcfe3f60 : 0xffffff801089caee 
0xffffff91fcfe3fb0 : 0xffffff80102a6af7 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOHIDFamily(2.0)[EF97107C-5B48-3B48-A1CF-D9BAC7C2632D]@0xffffff7f9179a000->0xffffff7f91811fff
            dependency: com.apple.driver.AppleFDEKeyStore(28.30)[B9C62BFE-2AF9-33FF-85AD-B2BCEEF35E57]@0xffffff7f9178f000

BSD process name corresponding to current thread: kernel_task

Mac OS version:
16A323

Kernel version:
Darwin Kernel Version 16.0.0: Mon Aug 29 17:56:20 PDT 2016; root:xnu-3789.1.32~3/RELEASE_X86_64
Kernel UUID: 622D2470-C34D-31F9-A62B-6AA9A3C6A3CD
Kernel slide:     0x0000000010000000
Kernel text base: 0xffffff8010200000
__HIB  text base: 0xffffff8010100000
System model name: MacBookPro11,3 (Mac-2BD1B31983FE1663)

System uptime in nanoseconds: 35541954432
last loaded kext at 24730380498: com.mice.driver.Wireless360Controller	1.0.0d15 (addr 0xffffff7f937bd000, size 20480)
loaded kexts:
com.mice.driver.Wireless360Controller	1.0.0d15
com.mice.driver.WirelessGamingReceiver	1.0.0d15
com.apple.driver.AppleHWSensor	1.9.5d0
com.apple.driver.AudioAUUC	1.70
com.apple.filesystems.autofs	3.0
com.apple.driver.AGPM	110.23.7
com.apple.driver.ApplePlatformEnabler	2.7.0d0
com.apple.driver.X86PlatformShim	1.0.0
com.apple.filesystems.ntfs	3.13
com.apple.driver.AppleOSXWatchdog	1
com.apple.driver.AppleGraphicsDevicePolicy	3.13.60
com.apple.driver.AppleUpstreamUserClient	3.6.4
com.apple.driver.pmtelemetry	1
com.apple.iokit.IOUserEthernet	1.0.1
com.apple.driver.AppleHDAHardwareConfigDriver	276.26
com.apple.GeForce	10.1.2
com.apple.iokit.IOBluetoothSerialManager	5.0.0f18
com.apple.driver.AppleHDA	276.26
com.apple.driver.AppleIntelHD5000Graphics	10.1.8
com.apple.Dont_Steal_Mac_OS_X	7.0.0
com.apple.driver.AppleBacklight	170.9.10
com.apple.driver.ACPI_SMC_PlatformPlugin	1.0.0
[... 130 lines cut ...]
Serial ATA Device: APPLE SSD SM0512F, 500.28 GB
Model: MacBookPro11,3, BootROM MBP112.0138.B17, 4 processors, Intel Core i7, 2.5 GHz, 16 GB, SMC 2.19f12
Network Service: Wi-Fi, AirPort, en0
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
Graphics: NVIDIA GeForce GT 750M, NVIDIA GeForce GT 750M, PCIe, 2048 MB
Model: MacBookPro11,3, BootROM MBP112.0138.B17, 4 processors, Intel Core i7, 2.5 GHz, 16 GB, SMC 2.19f12
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
Graphics: NVIDIA GeForce GT 750M, NVIDIA GeForce GT 750M, PCIe, 2048 MB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (7.21.171.10.1a16)
Bluetooth: Version 5.0.0f18, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0512F, 500.28 GB
USB Device: USB 3.0 Bus
USB Device: USB5534B
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: iPhone
USB Device: USB2134B
USB Device: Keyboard Hub
USB Device: Apple Keyboard
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1
```

FranticRain, 2016-09-24:
Cannot repro.

FranticRain, 2016-09-24:
I have a possibly related kernel panic in that the computer panics when I close the lid on my mac after using the driver. That sounds an awful lot like "Apple's new USB stack continues to be a broken garbage fire and we should really just throw gasoline on macOS and burn it to the ground and revert to 10.10."

christ776, 2016-09-25:
Thanks a lot for the feedback ,I actually came back to mention that I was closing the lid, then immediately disconnecting the dongle only to realize that when I re-opened the lid (usually the next day) the mac had already restarted because of the kernel panic.

skunkworker, 2016-09-25:
I also have gotten a kernel panic from this on sierra as well.

goldorak84, 2016-10-09:
Kernel panic on my side too : `Anonymous UUID:       2D281FF2-7201-0779-AA13-AA2E123F2269

Sun Oct  9 13:25:06 2016

*** Panic Report ***
panic(cpu 2 caller 0xffffff8007a09f4a): Kernel trap at 0xffffff7f8a7dc8be, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x000000000afcc000, CR4: 0x00000000001627e0
RAX: 0x0000000000000000, RBX: 0x0000000000000000, RCX: 0x0000000009000000, RDX: 0xffffff800806ab58
RSP: 0xffffff91136a3c70, RBP: 0xffffff91136a3cd0, RSI: 0x0000000001150002, RDI: 0x0000000000000000
R8:  0xffffff802a34e700, R9:  0x0000000000000001, R10: 0xffffff8008066fb0, R11: 0xffffff8007ebff80
R12: 0xffffff8012f59500, R13: 0x0000000000000000, R14: 0x0000000000000000, R15: 0xffffff8033cfe600
RFL: 0x0000000000010282, RIP: 0xffffff7f8a7dc8be, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x2, PL: 0

Backtrace (CPU 2), Frame : Return Address
0xffffff91136a3900 : 0xffffff80078f748c 
0xffffff91136a3980 : 0xffffff8007a09f4a 
0xffffff91136a3b60 : 0xffffff80078a9743 
0xffffff91136a3b80 : 0xffffff7f8a7dc8be 
0xffffff91136a3cd0 : 0xffffff7f8a7dc70d 
0xffffff91136a3d10 : 0xffffff7f887bee88 
0xffffff91136a3d90 : 0xffffff7f88725f9d 
0xffffff91136a3e10 : 0xffffff8007ec00ba 
0xffffff91136a3e80 : 0xffffff7f887258dd 
0xffffff91136a3ed0 : 0xffffff7f8872a7e1 
0xffffff91136a3f10 : 0xffffff7f8872a53f 
0xffffff91136a3f30 : 0xffffff8007ebd621 
0xffffff91136a3f80 : 0xffffff8007ebcc06 
0xffffff91136a3fb0 : 0xffffff80078a6af7 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOUSBHostFamily(1.1)[6A671CD8-5527-3A10-8675-1421D158D7A7]@0xffffff7f886ef000->0xffffff7f88756fff
            dependency: com.apple.driver.AppleBusPowerController(1.0)[DB526B45-1A45-3A81-A0C1-57F826CADEDF]@0xffffff7f886e2000
         com.apple.iokit.IOUSBFamily(900.4.1)[8F6207EC-608D-373A-B35E-E6578202F58D]@0xffffff7f88793000->0xffffff7f8882bfff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[731443D8-78D5-30C8-939A-1ED3E857CA22]@0xffffff7f88132000
            dependency: com.apple.iokit.IOUSBHostFamily(1.1)[6A671CD8-5527-3A10-8675-1421D158D7A7]@0xffffff7f886ef000
         com.mice.driver.WirelessGamingReceiver(1.0d15)[1A9C280E-FDE6-344B-AB32-1F3E6EB7FB06]@0xffffff7f8a7da000->0xffffff7f8a7e4fff
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[EF97107C-5B48-3B48-A1CF-D9BAC7C2632D]@0xffffff7f8885d000
            dependency: com.apple.iokit.IOUSBFamily(900.4.1)[8F6207EC-608D-373A-B35E-E6578202F58D]@0xffffff7f88793000

BSD process name corresponding to current thread: kernel_task

Mac OS version:
16A323

Kernel version:
Darwin Kernel Version 16.0.0: Mon Aug 29 17:56:20 PDT 2016; root:xnu-3789.1.32~3/RELEASE_X86_64
Kernel UUID: 622D2470-C34D-31F9-A62B-6AA9A3C6A3CD
Kernel slide:     0x0000000007600000
Kernel text base: 0xffffff8007800000
__HIB  text base: 0xffffff8007700000
System model name: iMac14,1 (Mac-031B6874CF7F642A)

System uptime in nanoseconds: 1394926928289
last loaded kext at 1336946904011: com.mice.driver.Wireless360Controller	1.0.0d15 (addr 0xffffff7f8a7e5000, size 20480)
last unloaded kext at 328340510798: com.apple.filesystems.msdosfs	1.10 (addr 0xffffff7f8a7d2000, size 61440)
loaded kexts:
com.mice.driver.Wireless360Controller	1.0.0d15
com.mice.driver.WirelessGamingReceiver	1.0.0d15
com.apple.driver.AppleBluetoothMultitouch	92
com.apple.driver.AudioAUUC	1.70
com.apple.filesystems.ntfs	3.13
com.apple.driver.X86PlatformShim	1.0.0
com.apple.driver.AGPM	110.23.7
com.apple.driver.ApplePlatformEnabler	2.7.0d0
com.apple.driver.pmtelemetry	1
com.apple.iokit.IOUserEthernet	1.0.1
com.apple.iokit.IOBluetoothSerialManager	5.0.0f18
com.apple.Dont_Steal_Mac_OS_X	7.0.0
com.apple.filesystems.autofs	3.0
com.apple.driver.AppleMikeyHIDDriver	127
[... 107 lines cut ...]
Model: iMac14,1, BootROM IM141.0118.B13, 4 processors, Intel Core i5, 2,7 GHz, 8 GB, SMC 2.14f24
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343531533641465238412D50422020
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343531533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x111), Broadcom BCM43xx 1.0 (7.21.171.10.1a16)
Bluetooth: Version 5.0.0f18, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
PCI Card: pci1b21,612, AHCI Controller, Thunderbolt@195,0,0
Serial ATA Device: APPLE HDD HTS541010A9E662, 1 TB
Serial ATA Device: SanDisk SDSSDHP256G, 256,06 GB
USB Device: USB 3.0 Bus
USB Device: Backup+ Desk
USB Device: USB3.0 Hub
USB Device: USB3.0 Hub
USB Device: Expansion Desk
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB2.0 Hub
USB Device: Vendor-Specific Device
USB Device: Intuos PS
USB Device: USB2.0 Hub
Thunderbolt Bus: iMac, Apple Inc., 23.10
Thunderbolt Device: Rugged mini, LaCie, 3, 5.4
`

christ776, 2016-10-14:
Still happens

```
Anonymous UUID:       5C49430D-212B-8D84-6799-C73E6F2F0EF3

Fri Oct 14 09:42:48 2016

*** Panic Report ***
panic(cpu 2 caller 0xffffff8007c9d79d): "OSMetaClass::_RESERVEDOSMetaClass6 called"@/Library/Caches/com.apple.xbs/Sources/xnu/xnu-3789.20.43/libkern/c++/OSMetaClass.cpp:315
Backtrace (CPU 2), Frame : Return Address
0xffffff9238673d80 : 0xffffff80076f348c 
0xffffff9238673e00 : 0xffffff8007c9d79d 
0xffffff9238673e10 : 0xffffff7f87f0932c 
0xffffff9238673e30 : 0xffffff8007c968de 
0xffffff9238673e80 : 0xffffff8007c97dca 
0xffffff9238673f10 : 0xffffff8007c97c36 
0xffffff9238673f60 : 0xffffff8007c9964e 
0xffffff9238673fb0 : 0xffffff80076a2af7 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOHIDFamily(2.0)[58B2294A-AD19-32D8-8CB7-B666C55FF1D4]@0xffffff7f87f06000->0xffffff7f87f7efff
            dependency: com.apple.driver.AppleFDEKeyStore(28.30)[B9C62BFE-2AF9-33FF-85AD-B2BCEEF35E57]@0xffffff7f87efb000

BSD process name corresponding to current thread: kernel_task

Mac OS version:
16B2327e

Kernel version:
Darwin Kernel Version 16.1.0: Mon Sep 12 19:36:59 PDT 2016; root:xnu-3789.20.43~78/RELEASE_X86_64
Kernel UUID: F983CF22-AA54-37E4-873A-153D6CA8059F
Kernel slide:     0x0000000007400000
Kernel text base: 0xffffff8007600000
__HIB  text base: 0xffffff8007500000
System model name: MacBookPro11,3 (Mac-2BD1B31983FE1663)

System uptime in nanoseconds: 3392554678117
last loaded kext at 1873144914499: com.apple.driver.AppleBluetoothMultitouch	92 (addr 0xffffff7f8ac47000, size 57344)
last unloaded kext at 1934339168172: com.apple.driver.usb.cdc	5.0.0 (addr 0xffffff7f8abde000, size 28672)
loaded kexts:
com.mice.driver.Wireless360Controller	1.0.0d15
com.mice.driver.WirelessGamingReceiver	1.0.0d15
com.apple.driver.AppleBluetoothMultitouch	92
com.apple.driver.AppleHWSensor	1.9.5d0
com.apple.driver.AudioAUUC	1.70
com.apple.filesystems.autofs	3.0
com.apple.driver.AGPM	110.23.11
com.apple.driver.ApplePlatformEnabler	2.7.0d0
com.apple.driver.X86PlatformShim	1.0.0
com.apple.filesystems.ntfs	3.13
com.apple.driver.AppleOSXWatchdog	1
com.apple.driver.AppleGraphicsDevicePolicy	3.13.69
com.apple.driver.AppleHDA	278.15
com.apple.driver.pmtelemetry	1
com.apple.driver.AppleUpstreamUserClient	3.6.4
com.apple.iokit.IOUserEthernet	1.0.1
com.apple.GeForce	10.1.2
com.apple.driver.AppleThunderboltIP	3.0.8
com.apple.driver.AppleIntelHD5000Graphics	10.2.0
com.apple.iokit.IOBluetoothSerialManager	5.0.1d14
com.apple.iokit.BroadcomBluetoothHostControllerUSBTransport	5.0.1d14
com.apple.driver.AppleLPC	3.1
com.apple.driver.AppleSMCLMU	208
com.apple.driver.AppleMuxControl	3.13.69
com.apple.driver.AppleCameraInterface	5.57.0
com.apple.driver.AppleMCCSControl	1.2.13
com.apple.driver.AppleIntelFramebufferAzul	10.2.0
com.apple.Dont_Steal_Mac_OS_X	7.0.0
com.apple.driver.AppleHV	1
com.apple.driver.AppleIntelSlowAdaptiveClocking	4.0.0
com.apple.driver.AppleUSBCardReader	404.1.1
[... 87 lines cut ...]
com.apple.driver.AppleACPIPlatform	5.0
com.apple.iokit.IOPCIFamily	2.9
com.apple.iokit.IOACPIFamily	1.4
com.apple.kec.Libm	1
com.apple.kec.pthread	1
com.apple.kec.corecrypto	1.0
Model: MacBookPro11,3, BootROM MBP112.0138.B17, 4 processors, Intel Core i7, 2.5 GHz, 16 GB, SMC 2.19f12
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
Graphics: NVIDIA GeForce GT 750M, NVIDIA GeForce GT 750M, PCIe, 2048 MB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (7.21.171.40.1a2)
Bluetooth: Version 5.0.1d14, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0512F, 500.28 GB
USB Device: USB 3.0 Bus
USB Device: USB5534B
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB2134B
USB Device: Keyboard Hub
USB Device: Apple Keyboard
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1
```

larianstudios, 2016-10-18:
Can you guys give me a radar number of your ticket?

Rhymzz, 2016-10-19:
Hey guys, i read your post cuz i was looking for this backtrace code.
I got the same problem as you have. Also updating El Capitan to Siera, during the instalation i got this freakin' kernel panic, com.apple.iokit.IOHIDFamily(2.0)[EF97107C-5B48-3B48-A1CF-D9BAC7C2632D].

I called Apple Netherlands, they couldn't help me further.
Do you guys have some more tricks or ideas?

skunkworker, 2016-10-20:
Nothing so far. I haven't found a way to use it on sierra.

Spidey002, 2016-10-21:
I've been having this issue as well. Just updated to macOS 10.12.1 public beta 5 (16B2553a) and the last time I unplugged the wireless dongle and my MacBook Pro slept, it did NOT crash. 
Going to try it again tonight.

Spidey002, 2016-10-22:
Played again last night with my wireless 360 controller (0.16.4 driver). No crash!
macOS 10.12.1 public beta 5 works with the driver. 
Hopefully, they'll be rolling out the public release soon. Hang in there everyone!

FranticRain, 2016-10-22:
Meanwhile, Apple insists nothing is wrong with the new USB implementation. ¯\_(ツ)_/¯

Rhymzz, 2016-10-22:
I installed yesterday sierra on a 32 go USB stick, everything went good.

christ776, 2016-10-23:
Hi ! 
I've installed macOS 10.12.1 Beta 5 yesterday, and still having the same issue.. 

```
Anonymous UUID:       5C49430D-212B-8D84-6799-C73E6F2F0EF3

Sat Oct 22 19:48:05 2016

*** Panic Report ***
panic(cpu 3 caller 0xffffff80100065ea): Kernel trap at 0xffffff801049e077, type 13=general protection, registers:
CR0: 0x000000008001003b, CR2: 0x000000010fdfa000, CR3: 0x0000000013bd9000, CR4: 0x00000000001627e0
RAX: 0xc0ffeec7ca73f186, RBX: 0xffffff804999e800, RCX: 0xffffff80308ac0e0, RDX: 0xffffff80306af040
RSP: 0xffffff923e383e10, RBP: 0xffffff923e383e10, RSI: 0x0000000000000000, RDI: 0xffffff804999e800
R8:  0x000000000000ffff, R9:  0x0000000000010001, R10: 0xffffff8010666fb0, R11: 0xffffff801053b584
R12: 0xffffff80308ac128, R13: 0xffffff80308ac0e0, R14: 0xffffff80306af040, R15: 0xffffff80308ac0e0
RFL: 0x0000000000010246, RIP: 0xffffff801049e077, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x000000010fdfa000, Error code: 0x0000000000000000, Fault CPU: 0x3, PL: 0

Backtrace (CPU 3), Frame : Return Address
0xffffff81ee7fddf0 : 0xffffff800fef368c 
0xffffff81ee7fde70 : 0xffffff80100065ea 
0xffffff81ee7fe050 : 0xffffff800fea5743 
0xffffff81ee7fe070 : 0xffffff801049e077 
0xffffff923e383e10 : 0xffffff7f913a02fc 
0xffffff923e383e30 : 0xffffff80104971be 
0xffffff923e383e80 : 0xffffff80104986aa 
0xffffff923e383f10 : 0xffffff8010498516 
0xffffff923e383f60 : 0xffffff8010499f2e 
0xffffff923e383fb0 : 0xffffff800fea2af7 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOHIDFamily(2.0)[1A48D920-280E-36FA-8D48-49B79A5656E6]@0xffffff7f9139d000->0xffffff7f91415fff
            dependency: com.apple.driver.AppleFDEKeyStore(28.30)[EA5D0966-E8EA-337A-98EB-195806E8F723]@0xffffff7f91392000

BSD process name corresponding to current thread: kernel_task

Mac OS version:
16B2553a

Kernel version:
Darwin Kernel Version 16.1.0: Mon Oct  3 14:03:15 PDT 2016; root:xnu-3789.21.3~20/RELEASE_X86_64
Kernel UUID: C6654CCD-B75E-3A86-8F3A-C4808664317C
Kernel slide:     0x000000000fc00000
Kernel text base: 0xffffff800fe00000
__HIB  text base: 0xffffff800fd00000
System model name: MacBookPro11,3 (Mac-2BD1B31983FE1663)

System uptime in nanoseconds: 3573932320055
last loaded kext at 245753804136: com.apple.filesystems.msdosfs	1.10 (addr 0xffffff7f933e4000, size 69632)
last unloaded kext at 350078574121: com.apple.filesystems.msdosfs	1.10 (addr 0xffffff7f933e4000, size 61440)
loaded kexts:
com.mice.driver.Wireless360Controller	1.0.0d15
com.mice.driver.WirelessGamingReceiver	1.0.0d15
com.apple.driver.AppleHWSensor	1.9.5d0
com.apple.driver.AudioAUUC	1.70
com.apple.driver.AGPM	110.23.11
com.apple.driver.ApplePlatformEnabler	2.7.0d0
com.apple.driver.X86PlatformShim	1.0.0
com.apple.filesystems.ntfs	3.13
com.apple.filesystems.autofs	3.0
com.apple.driver.AppleOSXWatchdog	1
com.apple.driver.AppleGraphicsDevicePolicy	3.13.74
com.apple.driver.AppleHDA	278.23
com.apple.driver.AppleUpstreamUserClient	3.6.4
com.apple.GeForce	10.1.2
com.apple.driver.pmtelemetry	1
com.apple.iokit.IOUserEthernet	1.0.1
com.apple.driver.AppleIntelHD5000Graphics	10.2.0
com.apple.iokit.IOBluetoothSerialManager	5.0.1f7
com.apple.Dont_Steal_Mac_OS_X	7.0.0
com.apple.iokit.BroadcomBluetoothHostControllerUSBTransport	5.0.1f7
````

Spidey002, 2016-10-28:
Welp, my Mac had crashed again when I tried to turn it on this morning. 
How many of you are using Steam?
Seems that the thing I did differently this time was that I left Steam running instead of quitting. 
I don't know.

Spidey002, 2016-11-10:
Still crashing on macOS 10.12.2 public beta 2

[Crash report 11-10-16.txt](https://github.com/360Controller/360Controller/files/583642/Crash.report.11-10-16.txt)

JohannesHoffmann, 2017-01-28:
+1 Appears after disconnecting the wireless dongle or a wired controller after a random time. I use Sierra and a xBox 360 and one controller.

masonjames, 2017-01-31:
After updating to 10.12.4 beta 1 today I immediately began to experience this issue.  Kernel panic led to being logged out. Happened about 6x today with no controller in use. Uninstalling this and restarting the computer returned me to stable usage

mele113, 2017-02-01:
Since I have upgraded to Sierra in November from ElCapitan the issue had arose with many similar Kernel Panics on my Mac. My work around was to use System Preferences app to turn 360 controller Off, prior to unplugging the USB plug which STOPPED all Panics. Its not a fix, but my only help

Spidey002, 2017-02-01:
That's the best suggestion I've heard yet. I'll give it a try next time I play.

GeneralMaster, 2017-02-05:
I have kernel panics as well. I boiled it down to the 360Controller kext.
I use the wireless version of the xbox 360 controllers on MacOS Sierra 10.12.3 on a late 2016 Macbook Pro 15''.
I too can confirm that the Kernel Panics go away, if I unload the kernel extensions for 360Controller on the terminal with
`sudo kextunload -b com.mice.driver.WirelessGamingReceiver
sudo kextunload -b com.mice.driver.Wireless360Controller`

This results is though that the USB wireless receiver doesn't turn on (the green light lights up), if you re-plug it in, until you do a
`sudo kextload -b com.mice.driver.WirelessGamingReceiver
sudo kextload -b com.mice.driver.Wireless360Controller`

While this is a workaround that kind of works, I also experienced some random kernel panics WHILE I used the controller in a game.
Plus, I sometimes forget to unload the kernel extension, before I put the laptop to sleep, which usually happens quickly after I stopped playing, and upon opening the lid of my Macbook, I'm being greeted with the white Apple logo and the laptop starting up again. So it crashes even when in sleep mode somehow.

Would be really great to soon have a fix for that.

christ776, 2017-02-14:
My workaround here : do not disconnect the dongle before closing the lid. After having closed the lid, if I disconnect the dongle, I dont get a kernel panic.

Spidey002, 2017-02-14:
I must've done this sometimes. Explains why sometimes I get the crash and sometimes I don't.

Cruplezone, 2017-04-01:
Yes this is still happening to me on 10.12.4 sadly. Possible for frantic to take a look or no?

FranticRain, 2017-04-17:
I can't promise anything, but the total rewrite of this driver for the new macOS USB strack that has been floating around for a while _should_ fix this, but it currently doesn't support, you guessed it, wireless 360 controllers. I've never been particularly well versed in that aspect of the code. Its entirely possible that @MaddTheSane has some insight on this, but honestly I can't promise anything. I really only know the wired stuff, and I made sure that part worked.

Cruplezone, 2017-04-18:
@FranticRain alright fair enough I've heard before you're not all that knowledgeable on the wireless side of things. As usb ports continue to be removed though in newer macs, having a single receiver proves to be really helpful for multiplayer. Hopefully you can look into the wireless side of things with this new version. Do you own a wireless controller to test with? I love your work, it's just sad to see a big part of it be very buggy. I'm sure we'd be willing to crowdfund you one off ebay or something. 

Just to let you know there's also some issue where disabling the driver doesn't disable wireless controllers, the Enable Driver box always goes unticked once you close out of system settings, and today I had a kernel panic by toggling the enable driver button without even going to sleep (it's disabled on my mac now). 

Keep on improving the project though a lot of us are thankful and appreciative of what you do!

FranticRain, 2017-04-18:
The enable driver tool is a hack at best, and a mess at worst. It actually doesn't do anything about wireless devices. All disable does is forcibly unload the "360Controller.kext" from kernel space. This _shouldn't_ impact the wireless devices at all because it doesn't touch the wireless device's kexts.

I do have a wireless dongle and controller, and I have run my head up against this before and couldn't figure it out. I wish I knew how to fix it. If I remember properly, the whole thing connects and runs, but you don't get any button data out of it. It just has to be the part I know the least about.

Cruplezone, 2017-04-20:
Ah a shame. Thought it was an issue with limited ability for testing, was gonna offer to send you a spare. This sadly will have to remain an uninstall then for me. Because even without sleeping, random kernel panics/reboots occur after unplugging the wireless dongle after use (after a couple of hours sometimes) which I really can't have happening.

Hopefully you take another shot at the wireless sometime , i'll be looking out. As I'm sure many other users are.

Spidey002, 2017-04-20:
The wireless controller connects and works. The issue is that after I’m done playing, and I disconnect the dongle, the computer crashes after going to sleep.

~ Spidey002

dapetcu21, 2017-05-12:
@christ776 I get the opposite behaviour. If I disconnect the dongle when the lid is closed, I get a KP when I wake it up from sleep. If I disconnect it with the computer on, no issues.

```
Anonymous UUID:       3EAAE04D-A530-CFBA-33C5-437D849A1EE8

Fri May 12 09:13:06 2017

*** Panic Report ***
panic(cpu 0 caller 0xffffff802f2011bd): Kernel trap at 0xffffff802f69ce17, type 13=general protection, registers:
CR0: 0x000000008001003b, CR2: 0x00000001013d8000, CR3: 0x000000003310b000, CR4: 0x00000000001626e0
RAX: 0xc0ffee34a63293f2, RBX: 0xffffff805d4dc800, RCX: 0xffffff805ec8e8c0, RDX: 0xffffff805022d0e0
RSP: 0xffffff9234a7be10, RBP: 0xffffff9234a7be10, RSI: 0x0000000000000000, RDI: 0xffffff805d4dc800
R8:  0x000000000000ffff, R9:  0x0000000000010001, R10: 0xffffff802f8671b0, R11: 0xffffff802f73a814
R12: 0xffffff805ec8e908, R13: 0xffffff805ec8e8c0, R14: 0xffffff805022d0e0, R15: 0xffffff805ec8e8c0
RFL: 0x0000000000010246, RIP: 0xffffff802f69ce17, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0x00000001013d8000, Error code: 0x0000000000000000, Fault CPU: 0x0, PL: 0, VF: 0

Backtrace (CPU 0), Frame : Return Address
0xffffff81edb3fc50 : 0xffffff802f0ec3ec 
0xffffff81edb3fcd0 : 0xffffff802f2011bd 
0xffffff81edb3feb0 : 0xffffff802f09d543 
0xffffff81edb3fed0 : 0xffffff802f69ce17 
0xffffff9234a7be10 : 0xffffff7faf90b820 
0xffffff9234a7be30 : 0xffffff802f695dee 
0xffffff9234a7be80 : 0xffffff802f6972da 
0xffffff9234a7bf10 : 0xffffff802f697146 
0xffffff9234a7bf60 : 0xffffff802f698b5e 
0xffffff9234a7bfb0 : 0xffffff802f09a8f7 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOHIDFamily(2.0)[BB7E26B3-36DF-3BB1-B09D-C8496FE63DFE]@0xffffff7faf908000->0xffffff7faf981fff
            dependency: com.apple.driver.AppleFDEKeyStore(28.30)[EA5D0966-E8EA-337A-98EB-195806E8F723]@0xffffff7faf8fd000

BSD process name corresponding to current thread: kernel_task

Mac OS version:
16E195

Kernel version:
Darwin Kernel Version 16.5.0: Fri Mar  3 16:52:33 PST 2017; root:xnu-3789.51.2~3/RELEASE_X86_64
Kernel UUID: 06F94FD7-451F-34A1-B13C-D68FF7EDE0A0
Kernel slide:     0x000000002ee00000
Kernel text base: 0xffffff802f000000
__HIB  text base: 0xffffff802ef00000
System model name: MacBookPro10,1 (Mac-C3EC7CD22292981F)

System uptime in nanoseconds: 92679089024
last loaded kext at 59163594242: com.mice.driver.Wireless360Controller	1.0.0d15 (addr 0xffffff7fb297d000, size 20480)
loaded kexts:
com.mice.driver.Wireless360Controller	1.0.0d15
com.mice.driver.WirelessGamingReceiver	1.0.0d15
org.virtualbox.kext.VBoxNetAdp	5.0.6
org.virtualbox.kext.VBoxNetFlt	5.0.6
org.virtualbox.kext.VBoxUSB	5.0.6
com.intel.kext.intelhaxm	6.0.1
org.virtualbox.kext.VBoxDrv	5.0.6
com.Cycling74.driver.Soundflower	2
com.intel.driver.EnergyDriver	2.0
com.apple.driver.AppleHWSensor	1.9.5d0
com.apple.driver.AGPM	110.23.17
com.apple.driver.ApplePlatformEnabler	2.7.0d0
com.apple.driver.X86PlatformShim	1.0.0
com.apple.filesystems.autofs	3.0
com.apple.filesystems.ntfs	3.13
com.apple.driver.AppleMikeyHIDDriver	131
com.apple.driver.AudioAUUC	1.70
com.apple.driver.AppleGraphicsDevicePolicy	3.14.41
com.apple.driver.AppleHDAHardwareConfigDriver	279.48
com.apple.driver.AppleHDA	279.48
com.apple.driver.pmtelemetry	1
com.apple.iokit.IOUserEthernet	1.0.1
[... 105 lines cut ...]
com.apple.driver.usb.AppleUSBXHCI	1.1
com.apple.driver.AppleUSBHostMergeProperties	1.1
com.apple.driver.AppleEFINVRAM	2.1
com.apple.driver.AppleEFIRuntime	2.0
com.apple.iokit.IOHIDFamily	2.0.0
com.apple.iokit.IOSMBusFamily	1.1
com.apple.security.sandbox	300.0
com.apple.kext.AppleMatch	1.0.0d1
com.apple.driver.AppleKeyStore	2
com.apple.driver.AppleMobileFileIntegrity	1.0.5
com.apple.driver.AppleCredentialManager	1.0
com.apple.driver.KernelRelayHost	1
com.apple.iokit.IOUSBHostFamily	1.1
com.apple.driver.AppleBusPowerController	1.0
com.apple.driver.DiskImages	444.50.16
com.apple.iokit.IOStorageFamily	2.1
com.apple.iokit.IOReportFamily	31
com.apple.driver.AppleFDEKeyStore	28.30
com.apple.driver.AppleACPIPlatform	5.0
com.apple.iokit.IOPCIFamily	2.9
com.apple.iokit.IOACPIFamily	1.4
com.apple.kec.pthread	1
com.apple.kec.Libm	1
com.apple.kec.corecrypto	1.0
```

RobCheung888, 2017-05-21:
I got this problem too! After I unplug my xbox 360 wireless dongle, the macbook air just keep spinning the beachball

jedmonston, 2017-05-24:
+1. MacBook Pro 2016 (13 inch base model) crashes once asleep, after using the wireless driver

nagybal, 2017-06-28:
+1 iMac Late 2013

barefeg, 2017-08-22:
same problem in 10.12.6

jericson, 2017-08-29:
For what it's worth, I plugged my dongle into a Thunderbolt monitor and got a kernel panic every night when my Mac went to sleep mode. But plugging the dongle into a USB hub and didn't get a panic last night. Not sure if this helps, but there you go.

kabutar111, 2017-09-11:
Anyone working on this even?

nagybal, 2017-09-11:
Could someone from the Driver Developer Team update this ticket please?!

mele113, 2017-09-14:
I previously commented about these Kernel Panics, I have same exact issue however months ago I started using system preferences to turn off wireless x-box 360 controller, then unplugging the receiver. All Panics STOPPED....This isn't a fix but just a work around I use, I have not experienced a Panic in 2 months

Spidey002, 2017-09-14:
I have seen the same result if I take the batteries out of my controller (essentially turning it off) before I unplug my dongle.

~ Spidey

dfdevlin, 2017-11-25:
+1 on OS X 10.13.1. I'll disconnect the dongle, close the lid, and kernel panic occurs. I'll try the workaround listed above, but it' be great if someone could be working on this.

FranticRain, 2018-06-02:
An idea popped in to my head today. Can someone here try this build and let me know if it still causes kernel panics? Sorry about the zip, Apple doesn't let me upload dmgs or pkgs to comments.

[Install360Controller.pkg.zip](https://github.com/360Controller/360Controller/files/2064649/Install360Controller.pkg.zip)

METALR0CKER, 2018-06-07:
I have just installed this build. Earlier in the day I had about 3 Kernel Panics from my mac going to sleep. After installing this new build no panics have occurred yet, but I will keep monitoring it and see how it works over a few days.

benders, 2018-08-01:
Unfortunately did not fix the issue here, although it seemed to improve it? With the 0.16.5 driver I have what appears to be a 100% repro rate for panic on sleep if the controller isn't powered off first. With the posted driver I was able to sleep without trouble once, but failed after I unplugged the adapter, plugged it back in and tried to sleep the mac again. (Testing on MacOS 10.13.6)