360Controller/360Controller #358: Xbox 360 wired controller wont work

steakcheese, 2016-09-23:
Product ID:	0x0501
  Vendor ID:	0x0e6f
Just updated to OS X Sierra, tried using the 0.16.4 version of the 360controllers, but nothing. It doesn't recognize my controller. Any solution?

FranticRain, 2016-09-24:
The only current solution is to revert your operating system. I cannot reproduce this error, so I cannot solve it.

steakcheese, 2016-09-24:
@FranticRain to what version would you suggest me to revert my operating system to? 10.11?

FranticRain, 2016-09-24:
Since this is a wired Xbox 360 controller, 10.11 should be perfectly fine. There are some issues with third party Xbox One controllers that require a revert to 10.10, but this controller should work just fine on 10.11.

steakcheese, 2016-09-27:
@FranticRain successfully downgraded to 10.11.6, but still nothing. any suggestions?

FranticRain, 2016-09-27:
Make sure the controller works with either Windows or and Xbox 360 at this point. I want to make sure that the controller works in the first place before I go looking for an issue. This controller is in the Info.plist, so it should be matching up no problem if it works at all. 

Also I want to make sure of your definition of not working. Are you saying that if you open your System Preferences and go to the Xbox 360 Controllers preference pane, the controller does not register any inputs there?

steakcheese, 2016-09-27:
@FranticRain Yes, after the first time it didn't work I wanted to make sure the problem wasn't with the controller so I connected it with an xbox 360 and it worked fine.
The controller still appears in the System Report on the USB list, proving that it is connected to the computer, but when I go to the Xbox 360 Controllers preference pane, it doesn't even appear there. No device is found, even though it is connected. So yeah, the controller does not register any inputs there.

FranticRain, 2016-09-27:
Hmmm. Can you open up your "Console.app" found in Applications/Utilities, and then connect the controller. Look for any output that appears when you connect the controller. The driver should be attempting to pair with that controller just fine. There might be some useful output in the console though. Let me know if you see any. Do you have any other drivers on the computer that might be interfering with this one? Such as some other Xbox 360 controller driver?

steakcheese, 2016-09-27:
@FranticRain 
[image]
This is what I got. Hope it helps. But no, I don't have any driver. Had to erase my whole HD when I downgraded to 10.11.6, so my computer is clean.

FranticRain, 2016-09-27:
Okay, so I was going to give you a more verbose version of the driver, but Xcode 6.4 doesn't run on Sierra, so I can't build the old version of the driver. So I was going to give you the new version of the driver, but for some reason, Apple doesn't like my signing certificate right now. I'll try and get this fixed up, but just know that I'm working on it.

steakcheese, 2016-09-27:
@FranticRain Thats a bummer, but thank you! I'll be waiting for the solution, and just to make it clear though, I'm on El Capitan now, not Sierra anymore.

FranticRain, 2016-09-27:
Well, the fancy installer is screwed six ways to Sunday, so I guess we'll have to do this a little ghetto. Unzip this kext, place it in `/Library/Extensions` then run the terminal commands:
```sudo chown -R root:wheel /Library/Extensions/360Controller```
```sudo kextutil -nt /Library/Extensions/360Controller.kext```
``` sudo kextutil /Library/Extensions/360Controller.kext```

The first line takes ownership of the driver so it runs properly, the second lets you know if the driver has been signed properly. The third one should load the driver into the kernel and should work. If that third command gives you some trouble, then make sure all of your controllers are disconnected and run:
```sudo kextunload -b com.mice.driver.Xbox360Controller```
Then re-run that third command.

[360Controller.kext.zip](https://github.com/360Controller/360Controller/files/494479/360Controller.kext.zip)

If the driver attempts to pair with your controller at all, you should see "ENTERED START" in your Console.app.

I'm aware you're on El Cap, not Sierra. But I'm on Sierra, and I'm too busy with other stuff to care enough to revert just for supporting a driver I don't even use.

FranticRain, 2016-09-29:
Okay, I reverted to El Cap because some other stuff was busted in Sierra.

Here's an installer for a version with a little more information. Basically, just open up your console and then plug in your controller. Tell me if you see anything in the Console output.

https://drive.google.com/file/d/0B5PLCKoMd4zGTTNWbGIwc3FUWnc/view?usp=sharing

Dave-Melia, 2016-10-02:
This works a treat for me on MacOS Sierra.  Thank you for this, @FranticRain.

FranticRain, 2016-10-14:
Closing due to inactivity.

closed 2016-10-14 by FranticRain