360Controller/360Controller #359: Wireless Xbox 360 controllers won't connect

JohnGillanders, 2016-09-26:
Hi, 
Thanks for your great work on this project. My wireless Xbox 360 controllers (I have two) won't connect using the Xbox 360 wireless receiver for PC using this driver.
Key facts:
* It used to work - I think I upgraded Mac OS X since then (maybe that broke it...?)
* When it used to work, it was one of the versions when the project was hosted at: https://github.com/d235j/360Controller-n
* I am now on Mac OS X Yosemite 10.10.5 on a MacBook Pro 15" 2009 model
* I have uninstalled the old version of the driver and have reinstalled the latest version from this site (Version 0.16.4)
* The Xbox 360 Controllers preference pane appears fine
* When I plug in the receiver, the green light goes on
* After I switch on the controller and try to pair by pressing the button on the receiver (which starts flashing) and then pressing and holding the connect button on the back of the controller, what happens is that the light on the controller spins once (or twice sometimes) then all four lights around the circle flash on and off slowly (the receiver light becomes solid green). It just stays like this no matter how long I leave it or how many times I've tried (with each controller).
* I have tried restarting several times
* I have tried the instructions listed here: https://readme.highfidelity.com/docs/xbox-360-controller-on-mac-os-x-yosemite , which did not work for me
* I noticed that in the preference pane, under 'Advanced', the 'Enable driver' option is NOT ticked when I first installed the driver, and seems to become unticked after each restart of the computer. I have tried everything mentioned above after ticking 'enable driver', but it still does not work.
* I thought that my receiver was an official Microsoft one, but I see it has "XBOX 360" written on it and I see that ones listed as official have 'Microsoft' written on them, so maybe it's a counterfeit one. Would that cause it not to work now?
The older version of the driver used to work okay with this receiver.

Please help - I miss my gaming! :)

Am happy to buy an official Microsoft receiver if that's the issue.

Thanks a lot,
John

excessuk, 2016-12-04:
I have exactly the same issue, identical symptoms. This is the first time I try this driver so I can't attest that this has worked in the past for me. I'm running macOS Sierra 10.12.1.

JohnPaulHarold, 2017-01-12:
I'm seeing almost identically similar to this. I'm on El Capitan 10.11.2, and am using official 360 wireless controllers, with an equally official receiver. Oddly enough, the unofficial receiver I have works fine (I've bought an official receiver as I think it might solve some odd issues with companion software on Windows.) I can provide whatever codes, IDs will help, if with this driver is where the issue is likely to lie.

MartynF, 2017-05-30:
Also same problem. The controllers worked fine before an update to 10.12.4 after which they won't pair.

yang, 2018-11-24:
Maybe I misunderstood the README. 

```
Wireless Xbox 360 controllers are supported with the Xbox 360 Wireless Gaming Receiver. 
```

I'm using the wireless USB receiver. I'm interpreting the notice before this sentence as being about some other way of using wireless controllers directly without a USB receiver. Either way, I would appreciate being clued in on this. Thanks.

yang, 2018-11-24:
@FranticRain: That block was what I was referring to with "the notice before this sentence."

praveenvj1979, 2019-01-29:
How do we disable the driver? Do I need to uninstall and reinstall each time I need to use the controller. 

I checked in the System preference > and the enable drive check box is unchecked. But the XBOX 360 controller works even when it is unchecked

sylaryang, 2019-12-18:
The same situation happens to me. macOS Catalina 10.15.1. I used 0.16.11 driver.
 all four lights around the circle flash on and off slowly. What can I do? Any suggestions?

cystbear, 2020-01-14:
Take a look https://github.com/360Controller/360Controller#wireless-xbox-360-controllers note

```
As of macOS 10.11, Wireless Xbox 360 controller support causes kernel panics. 
This issue cannot be resolved with minor changes to the driver, and requires that the driver 
be re-written from scratch to resolve the issue. Due to an excess of caution, 
we have disabled Wireless Xbox 360 controller support as of 0.16.6. If you want 
to use a wireless controller, download 0.16.5 or earlier and disable the driver before 
the computer enters a "sleep" state in order to prevent kernel panics. Alternatively, 
you can revert to a macOS version before 10.11 to avoid this issue.
```

use [0.16.4](https://github.com/360Controller/360Controller/releases/tag/v0.16.4) version to handle this issue, works fine for me

tturan1907, 2020-06-03:
I don't know if this thread is still alive, but I have a very very similar issue here. 
Can't currently reach the wiki.highfidelity troubleshooting though since the page seems to be down now.

> use 0.16.4 version to handle this issue, works fine for me
@cystbear : Can you specify the hardware/software you're using? 
- What MacOS? 
- Wireless Receiver 3rd party or official?
- Do you know which releases before/after 0.16.4 work/doesn't work for you? OP seems to be running into this issue using 0.16.4, and says it worked fine with the older version. I tried several older versions to no avail here.

cystbear, 2020-06-04:
@tturan1907 
MacOS: 10.14.6
Wireless Receiver: official
Driver ver: 0.16.4

I am using `0.16.4` and everything fine for me.

samkern108, 2021-02-10:
> I am using `0.16.4` and everything fine for me.

@cystbear Are you disabling the driver as specified by the docs? It sounds like a headache but I don't want to run the risk of kernel shenanigans.

Jimmysikisken, 2023-05-05:
I was hopeful for this app as I have 4 Xbox 360 wireless controllers and a MBP running Catalina.

But this news made me give up and just go and buy the newer Xbox controllers that work with Catalina out of the box, without requiring the receiver dongle