360Controller/360Controller #365: When will a version that supports macOS Sierra 10.12 be released?

urbanman2004, 2016-10-11:
When will a version that supports macOS Sierra 10.12 be released? Currently installed 360ControllerInstall_0.16.4.dmg on 10.12 and it doesn't recognize my controller. However I got the drivers working in the same system in 10.11, El Capitan. Any helpful advice is appreciated. Thanks

sebastienkb, 2016-10-11:
You restarted a couple of times to get it working, right?

Could you try this? This allows to run applications "downloaded from anywhere" (disabled by default since Sierra).
`sudo spctl --master-disable`

If that doesn't work, please run this to get back to normal:
`sudo spctl --master-enable`

FranticRain, 2016-10-14:
Not as it currently stands. The build tools (both Packages and Xcode 6.4) don't work on Sierra. I think only Rodrigo and I have kext signing certs and I'm not upgrading the Sierra because it breaks stuff that I need. We can't build on newer Xcode unless we start using the elCap branch, which limits releases to only supporting elCap and Sierra, which means multiple versions of the driver. Additionally, the elCap branch doesn't work with wireless Xbox 360 controllers, so that's a BIT of an issue.

urbanman2004, 2016-10-19:
O ok. Thanks for the heads up... Guess I'll stick to El Cap for gaming for now

nbrunett, 2016-10-19:
Not sure if it helps but I've been able to use the 0.16.4 release with a Rock Candy 360 controller on Sierra. Feel free to ignore if this is getting out into the weeds.

FranticRain, 2016-10-19:
The current release *should* work with Sierra for wired controllers, but there are specifically issues with wireless 360 controllers causing a kernel panic.

oferhaz, 2016-10-25:
My wired (Original xbox) didn't show up in menu. I use a self-made USB adapter and I see the controller on system report when connected. macos sierra

FranticRain, 2016-10-25:
@oferhaz Did your controller work with a previous version of MacOS? If not, check out this: #319 #267

tracicot, 2016-10-26:
The driver works on my Sierra. Granted, I've only gotten it to work with one game so far. :(

jasarien, 2016-11-01:
I played a good couple of hours of Portal 2 on Sierra today. Driver seems to work perfectly with my Xbox Elite Controller.

urbanman2004, 2016-11-06:
I installed 10.12.1 and it is now working

mandersondesign, 2017-01-13:
My problem here is the buttons are like... sticking... with a microusb xbox one controller (newer edition). The thumbsticks work, and so do the triggers. It's like the analog is working just fine, but the digital inputs are slowed or distorted or something. Wish I hadn't updated to Sierra :/

Any thoughts on this issue?

roxas1403, 2017-07-01:
@urbanman2004 hi there, how did you get to connect your wireless controller? I'm currently running Mac OS X 10.12.2 (Sierra) and had tried the drivers 0.16.4 and 0.16.5, but my Xbox 360 wireless controller doesn't seem to connect to the receiver. Please help.

bfeinstein, 2017-07-25:
I am in the same boat as roxas1403. If I knew a thing about coding I would offer to help fix it. Running 10.12.6 FWIW.

scyhssm, 2017-12-12:
I have a same problem on 10.12.6.