360Controller/360Controller #368: Wired 360 controllers turn off.

jskybld, 2016-10-13:
Just installed the driver and my controllers work perfectly. My only issue is once I set my computer to sleep and come back a few hours later, I find the controllers have turned off (no lights) and don't turn on once the system wakes up. My workaround as of now is to disconnect and reconnect the controller altogether (which seems to mess up controller settings in Dolphin Emulator). I'm running El Capitan 10.11.6. Thank you to everyone contributing to this project, and apologies if this has been covered already!

FranticRain, 2016-10-14:
MacOS turns off power to USB ports when you sleep your computer. Not a whole lot we can do about that. If Dolphin doesn't recognize the controller properly on a reconnect, that's a Dolphin issue.

jskybld, 2016-10-14:
Thats understandable, but the issue is that the controllers dont turn back
on once my computer wakes up from sleep. Surely I'm not the only one who
puts their computer to sleep?

FranticRain, 2016-10-14:
Okay, I'll leave this one open with the note:

Power state settings, probably fixed with the Sierra kernel panic fix.