360Controller/360Controller #369: Wired Xbox One Controller not recognized

Opphitter16, 2016-10-13:
I have a wired xbox one controller I have been trying to connect. I just downloaded and installed the 0.16.4 release and I only get "no devices found" under my preferences pane. Please help! I'm hoping this isn't a killer but I am running OSX 10.11.6.

Opphitter16, 2016-10-13:
Also the controller does vibrate when I plug it in to the Mac. And I do have an entry for a controller in my system report. If it helps, here are the product and vendor IDs.

Product ID:	0x02d1
Vendor ID:	0x045e  (Microsoft Corporation)

FranticRain, 2016-10-14:
Uninstall other Xbox One controller drivers, such as Xone-OSX. Make sure to open the preference pane after you connect the controller. It sometimes has issues with live loading controllers. Does the controller illuminate at all?

Opphitter16, 2016-10-14:
Ok, I have uninstalled my other drivers and I made sure to open the
preference pane after I connected my controller but nothing has changed as
far as symptoms. I also uninstalled and reinstalled the 0.16.4 release
again.  My controller doesn't light up unless I press the button myself and
that doesn't seem to change anything for me. To clarify, my controller did
not light up before I had removed the other drivers either.

FranticRain, 2016-10-14:
After you plug in the controller, run this command in your terminal: `sudo kextstat`
It will print out a bunch of lines of data, and I need to see all of that.

iwatchhentaifortheplot, 2016-10-20:
Hey I think I'm having the same problem as this guy, im may not of uninstalled Xone-OSX but I'm not sure how too, but I copied what my terminal said after I put in the command

Index Refs Address            Size       Wired      Name (Version) <Linked Against>
    1   81 0xffffff7f80a3e000 0x8c50     0x8c50     com.apple.kpi.bsd (14.5.0)
    2    7 0xffffff7f80da3000 0x28c0     0x28c0     com.apple.kpi.dsep (14.5.0)
    3  104 0xffffff7f80a03000 0x20500    0x20500    com.apple.kpi.iokit (14.5.0)
    4  110 0xffffff7f80a24000 0xbf50     0xbf50     com.apple.kpi.libkern (14.5.0)
    5   96 0xffffff7f80a00000 0x2d50     0x2d50     com.apple.kpi.mach (14.5.0)
    6   46 0xffffff7f80a30000 0x8590     0x8590     com.apple.kpi.private (14.5.0)
    7   62 0xffffff7f80a39000 0x4900     0x4900     com.apple.kpi.unsupported (14.5.0)
    8    0 0xffffff7f80ddc000 0xc000     0xc000     com.apple.kec.pthread (1) <7 6 5 4 1>
    9    1 0xffffff7f80de8000 0x9000     0x9000     com.apple.kec.Libm (1) <4>
   10    3 0xffffff7f80f99000 0xa0000    0xa0000    com.apple.kec.corecrypto (1.0) <7 6 5 4 3 1>
   11   24 0xffffff7f81334000 0x9000     0x9000     com.apple.iokit.IOACPIFamily (1.4) <7 6 4 3>
   12   30 0xffffff7f80b24000 0x2f000    0x2f000    com.apple.iokit.IOPCIFamily (2.9) <7 6 5 4 3>
   13    2 0xffffff7f8291d000 0x5f000    0x5f000    com.apple.driver.AppleACPIPlatform (3.1) <12 11 7 6 5 4 3 1>
   14    1 0xffffff7f81039000 0xb000     0xb000     com.apple.driver.AppleFDEKeyStore (28.30) <10 7 6 5 4 3 1>
   15    2 0xffffff7f8132d000 0x7000     0x7000     com.apple.iokit.IOReportFamily (31) <5 4 3>
   16    7 0xffffff7f80a47000 0x26000    0x26000    com.apple.iokit.IOStorageFamily (2.0) <7 6 5 4 3 1>
   17    0 0xffffff7f81424000 0x19000    0x19000    com.apple.driver.DiskImages (398) <16 7 6 5 4 3 1>
   18    1 0xffffff7f82250000 0xf000     0xf000     com.apple.driver.AppleCredentialManager (1.0) <7 5 4 3 1>
   19    2 0xffffff7f80da9000 0xd000     0xd000     com.apple.driver.AppleMobileFileIntegrity (1.0.5) <7 6 5 4 3 2 1>
   20    0 0xffffff7f8225f000 0x23000    0x23000    com.apple.driver.AppleKeyStore (2) <19 18 10 7 6 5 4 3 1>
   21    0 0xffffff7f80da6000 0x2000     0x2000     com.apple.security.TMSafetyNet (8) <7 6 5 4 2 1>
   22    2 0xffffff7f80db6000 0x5000     0x5000     com.apple.kext.AppleMatch (1.0.0d1) <4 1>
   23    1 0xffffff7f80dbb000 0x17000    0x17000    com.apple.security.sandbox (300.0) <22 19 7 6 5 4 3 2 1>
   24    0 0xffffff7f80dd2000 0x9000     0x9000     com.apple.security.quarantine (3) <23 22 7 6 5 4 2 1>
   25    0 0xffffff7f82993000 0x8000     0x8000     com.apple.nke.applicationfirewall (161) <7 6 5 4 3 1>
   26    0 0xffffff7f828f0000 0x3000     0x3000     com.apple.driver.AppleAPIC (1.7) <4 3>
   27    3 0xffffff7f812ba000 0x4000     0x4000     com.apple.iokit.IOSMBusFamily (1.1) <5 4 3>
   28    0 0xffffff7f82982000 0x7000     0x7000     com.apple.driver.AppleACPIEC (3.1) <27 13 11 5 4 3>
   29    0 0xffffff7f82163000 0x4000     0x4000     com.apple.driver.AppleSMBIOS (2.1) <7 4 3>
   30    0 0xffffff7f82191000 0x8000     0x8000     com.apple.driver.AppleRTC (2.0) <11 5 4 3 1>
   31    5 0xffffff7f81044000 0x76000    0x76000    com.apple.iokit.IOHIDFamily (2.0.0) <14 7 6 5 4 3 2 1>
   32    0 0xffffff7f8298c000 0x4000     0x4000     com.apple.driver.AppleACPIButtons (3.1) <31 13 11 7 6 5 4 3 1>
   33    1 0xffffff7f82581000 0x5000     0x5000     com.apple.driver.AppleEFIRuntime (2.0) <7 6 5 4 3>
   34    1 0xffffff7f82586000 0xb000     0xb000     com.apple.driver.AppleEFINVRAM (2.0) <33 7 6 5 4 3 1>
   35    0 0xffffff7f82169000 0x8000     0x8000     com.apple.driver.AppleSmartBatteryManager (161.0.0) <27 11 5 4 3 1>
   36    0 0xffffff7f8250c000 0x3000     0x3000     com.apple.driver.AppleHPET (1.8) <11 7 5 4 3>
   37    1 0xffffff7f8228d000 0x8000     0x8000     com.apple.driver.AppleIntelLpssDmac (2.0.57) <12 11 5 4 3>
   39    1 0xffffff7f822ac000 0xa000     0xa000     com.apple.driver.AppleIntelLpssGspi (2.0.57) <12 11 7 5 4 3>
   40   10 0xffffff7f80f11000 0x65000    0x65000    com.apple.iokit.IOUSBFamily (720.4.4) <12 7 5 4 3 1>
   41    0 0xffffff7f810ec000 0x22000    0x22000    com.apple.driver.AppleUSBXHCI (710.4.11) <40 12 7 5 4 3 1>
   42    1 0xffffff7f822b6000 0x14000    0x14000    com.apple.driver.AppleIntelLpssSpiController (2.0.57) <39 37 11 5 4 3>
   43    1 0xffffff7f824df000 0x17000    0x17000    com.apple.driver.AppleHSSPISupport (41.3) <42 11 5 4 3 1>
   44    5 0xffffff7f80ed2000 0x2c000    0x2c000    com.apple.iokit.IONetworkingFamily (3.2) <7 6 5 4 3 1>
   45    1 0xffffff7f8165d000 0x7000     0x7000     com.apple.driver.mDNSOffloadUserClient (1.0.1b8) <44 4 3 1>
   46    1 0xffffff7f81664000 0xab000    0xab000    com.apple.iokit.IO80211Family (730.60) <44 10 7 6 5 4 3 1>
   47    0 0xffffff7f81717000 0x5e6000   0x5e6000   com.apple.driver.AirPort.Brcm4360 (930.37.3) <46 45 44 12 7 6 5 4 3 1>
   48    5 0xffffff7f811ab000 0xde000    0xde000    com.apple.iokit.IOThunderboltFamily (4.2.2) <5 4 3 1>
   49    0 0xffffff7f82078000 0x1b000    0x1b000    com.apple.driver.AppleThunderboltNHI (3.1.7) <48 12 11 5 4 3 1>
   50    2 0xffffff7f81fff000 0x24000    0x24000    com.apple.driver.AppleMultitouchDriver (264.6) <31 5 4 3 1>
   51    0 0xffffff7f824f9000 0x9000     0x9000     com.apple.driver.AppleHSSPIHIDDriver (41.3) <50 43 31 5 4 3 1>
[... cut ...]
  130    0 0xffffff7f827c6000 0x14000    0x14000    com.apple.driver.AGPM (110.19.6) <92 81 79 77 12 5 4 3>
  131    0 0xffffff7f821e9000 0x4000     0x4000     com.apple.driver.ApplePlatformEnabler (2.2.0d4) <7 5 4 3>
  132    0 0xffffff7f81f40000 0x5000     0x5000     com.apple.driver.AudioAUUC (1.70) <101 77 12 11 7 5 4 3 1>
  133    1 0xffffff7f81e80000 0x5000     0x5000     com.apple.kext.triggers (1.0) <7 6 5 4 3 1>
  134    0 0xffffff7f81e85000 0x9000     0x9000     com.apple.filesystems.autofs (3.0) <133 7 6 5 4 3 1>

FranticRain, 2016-10-21:
Assuming that your controller was plugged in when you ran `kextstat`, then no driver was loaded when your controller was plugged in. Was you controller plugged in when you ran this `kextstat`? It needs to be immediately after a connect, before the drivers have a chance to be unloaded. Assuming you have a supported controller and no other driver is matching to the controller, the driver will load. 

If you previously had Xone-OSX installed and aren't sure if you uninstalled it properly, then there is a 99% chance that you didn't uninstall it properly. The download for Xone-OSX is a DMG that contains two files. One of the files is an `uninstall.command`. All you have to do is double click on that script, and it will uninstall the driver from your system. Uninstall Xone-OSX and give it another shot.

FranticRain, 2016-10-21:
Okay, I need your controller's vendor and product IDs from ```Apple Menu -> System Preferences -> Xbox 360 Controllers```

iwatchhentaifortheplot, 2016-10-21:
It says No devices found

FranticRain, 2016-10-24:
Sorry, I gave you the wrong path there. Here it is from the README, "...make an issue with the "Product ID" and "Vendor ID" of the controller. These can be found by accessing the Apple menu, selecting "About this Mac", and then selecting "System Report..." on the "Overview" tab. On the left hand side of the new window, select the "USB" option under "Hardware". If the controller is plugged in, there should be an entry in this window called "Controller"."

iwatchhentaifortheplot, 2016-10-27:
Maybe theres a problem with my usb to micro-usb because when i plug it in there isnt a entry titled controller, only one titled Bluetooth USB Host Controller but i'm assuming thats not it as its still there regardless of if i have the controller plugged in or not

FranticRain, 2016-10-27:
Yeah, that means that you are using a "charge" cable, not a "data" cable. Some micro-usb cables don't have data hookups. And if we can't get data, we can't make a controller work like a controller.

iwatchhentaifortheplot, 2016-10-27:
okay thanks, i'll look for another cable to see if that one works

milanvanhorssen, 2016-12-05:
For me it says a mb per second count so i am using the right cable but my controller when i plug it in only vibrates and dusnt show a light

milanvanhorssen, 2016-12-05:
WAIT I FOUNd it. the controller is in there but it isnt working.
it just says controller
and is made by microsoft
[image]

moscoso, 2016-12-05:
I just wanted to verify what @FranticRain said made it work. I had the exact same issue as @iwatchhentaifortheplot and I just found another micro usb cable and as soon as I plugged it in it lit up. It could also be detected with the system preference pane open as well.

FranticRain, 2017-04-17:
Going to go ahead and close this one down. If your device isn't in that list, then you are using a charge cable and you need to change to a data cable. The VID and PID provided are already in the Info.plist, so it should work just fine. If you continue to have issues, make sure to update the firmware for your controller and try again.

closed 2017-04-17 by FranticRain

aj-21, 2018-03-30:
I am having the same issues. just trying to get controller to connect to plays games. have no knowledge with programming. the controller Product and Vendor ID is as follows: Product ID:	0x02dd
  Vendor ID:	0x045e  (Microsoft Corporation). When I pull up is recognized in system report but not in driver on system preference. Any help is appreciated. Thank You

twyns, 2018-04-06:
It says my controller is there with a vendor Id of 0x045e and product Id of 0x02d1

FranticRain, 2018-04-06:
@aj-21 @twyns

Please ensure that the driver has been allowed in System Preferences -> Security and Privacy -> General. This approval UI is only present in the Security & Privacy preferences pane for 30 minutes after the alert. Until you approve the kext, future load attempts will cause the approval UI to reappear but will not trigger another user alert.

[image]

Plzhelpm, 2018-04-13:
Help my controller is all plugged in and it still does not work

Plzhelpm, 2018-04-13:
I have done everything you said and it still does not turn on or respond to the computer

That-Radley-Guy, 2018-04-17:
I have done everything and  it says my controller is plugged in but it doesn't work.