360Controller/360Controller #374: Rock band 1 drums not recognized

oneally, 2016-10-27:
Hey all,

After searching for a solution for what seemed like days i'm thankful i found this driver update! Thanks so much!

I just installed the latest build 16.4 and everything went well with the installation. The preference pane see's the drum kit, But as an xbox 360 controller not the harmonix drum kit previously in the much older build. When i try to click on the dropdown menu to see if its there nothing happens. I'm trying to use the drum kit as a midi controller with gamepad companion in ableton. The system does recognize the device and here is the info. This is 1st version of rockband drums. 

Harmonix Drum Kit for Xbox 360:

  Product ID:	0x0003
  Vendor ID:	0x1bad
  Version:	1.02
  Serial Number:	00193F3C
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Harmonix Music
  Location ID:	0xfa131000 / 7
  Current Available (mA):	500
  Current Required (mA):	500

RodrigoCard, 2016-10-28:
Did you close this issue accidentally? Or did you still want this to be added?

oneally, 2016-10-28:
Hi Rodrigo,

I found a workaround, But if there is a current fix that would be great! Thanks for the quick reply!

FranticRain, 2016-10-30:
This isn't a new device. It is in the plist with the key "RockBandDrums"

RodrigoCard, 2016-11-01:
Oh, sorry about that

closed 2016-11-01 by RodrigoCard

rwreay, 2018-01-08:
@oneally, what's your workaround? I'm apparently having a simliar issue and would like to know what you did.