360Controller/360Controller #375: 3.5mm Jack on Xbox One S Controller

buffaloseven, 2016-10-29:
Not an issue, per say, but just wondering if it's within the scope of this driver to surface the Xbox One S controller's 3.5mm jack as an audio device to be able to port audio through it when plugged in via USB?

FranticRain, 2016-10-30:
#356

Short answer: No

Long answer: Its an incredibly niche problem given that if you are wired, you are usually close to your computer in the first place, so moving the audio to the controller, which has an inferior sound card to your Mac, doesn't do much for most people. This obviously isn't an absolute, and there are cases where people may want this. Secondly, no one is actively developing this driver anymore. Its pretty much purely maintenance at this point.

dsc, 2018-05-21:
I actually bought the XB1 S controller almost precisely because it was both bluetooth AND had an audio jack, but we all know the bloothpart is spotty at best, but more importantly, the 3.5 jack doesn't work in bluetooth mode. So I've ended up using it via USB cable, as I like the feel of the buttons over the wireless 360. I'm mostly sympathizing with OP, as I'm a little bummed the audio jack *never* works, but yeah who cares at the point where I'm tethered with a cable anyway.