360Controller/360Controller #382: MacBook Pro 15" late 2016 compatibility (USB-C)

klopfdreh, 2016-11-09:
Hey guys,

because I just ordered my new MacBook I just wondered if the current version of the 360Controller driver supports USB-C?

kind regards and thanks in advance

MaddTheSane, 2016-11-09:
I have used an Xbox One controller on my 2015 MacBook using the A/V USB adaptor with no issues.

klopfdreh, 2016-11-09:
Great to hear and thanks a lot for the fast response!

I think if this is confirmed by the devs too, this issue can be closed.

FranticRain, 2016-11-09:
Nothing about USB-C changes how this driver works. Its just another form factor for a serial connection. As long as you have a USB-C to Micro-B data cable or USB-C to USB-A female data cable, it will work.

klopfdreh, 2016-11-09:
Thanks a lot!

/close

closed 2016-11-09 by klopfdreh