360Controller/360Controller #383: Enable Driver Toggle bug

BrianMcElwain, 2016-11-11:
It seems that clicking the toggle to enable the driver asks for my OSX system password, but appears to toggle on/of anyway even if I simply cancel the authentication prompt without typing any password. Typing this out, I realised I should check if the also still controller enables/disables according to this toggle as such, but I am not on that machine right now anyway.

FranticRain, 2017-04-17:
From what I know, its hard to tell if the applescript that disables the driver knows if it has completed successfully or not. Definitely something someone should look into at some point.

BrianMcElwain, 2017-04-18:
Ah I have since sold my MacBookPro so I must abscond on this end.

FranticRain, 2019-10-19:
This should be fixed in alpha 4.

closed 2019-10-19 by FranticRain