360Controller/360Controller #385: Feature Investigation: Emulate XInput

SimplGy, 2016-11-14:
I'm trying to figure out how this project basically works. Does it:

* A: Take raw input from the USB bus and rebroadcast it as XInput by (naive example) mapping `00001000` to `yAxis: +1`?
* B: Do something completely different?

If it's A, **how large an effort would it be to support mapping any arbitrary gamepad's DirectInput input into XInput. Is this small? Medium? Large?**

I'm thinking of starting a fork for this, which you could merge back if you like.

If it works, no reason it couldn't support 3rd party 360 controllers too, since those only fail, I'm thinking, because they don't correctly implement the XInput spec.

> Background:
>
> I have tried a number of controller styles: Official Xbox 360, Buffalo Classic USB Gamepad, Steam Controller, PS4.
>
> For whatever reason, the controller that works best in the most games is the 360 controller. Even games with pretty advanced controller detection and mapping like "Lovers in a Dangerous Spacetime" can't handle configurations like two 360 controllers and one unknown controller.
>
> This makes me want to have all my controllers emulate 360 controllers.

MaddTheSane, 2016-11-14:
As XInput and DirectInput are Windows technologies, they are not available for Mac OS X.

This project:

1. Provides an HID descriptor so Mac OS X knows how to interpret the controller's input.
2. Provides a Force Feedback driver.
3. Provides a driver for the 360 wireless dongle and controller.

SimplGy, 2016-11-14:
Oh, I see. I thought you could still broadcast in that format.

So if I understand correctly, games that work with this driver are not receiving `XInput` messages, they are discovering a [HID](https://developer.apple.com/library/content/documentation/DeviceDrivers/Conceptual/HID/overview/overview.html#//apple_ref/doc/uid/TP40000970-CH203-CJAHEAFF) that this project makes available.

If this I think all gamepads on OSX must be HIDs, but they just broadcast in different formats--maybe the Y axis is on the 4th bit instead of the 3rd, that sort of thing?

So since this particular project works so well with so many games, the objective for me would be to allow you to trap all input from 3rd part HIDs and let folks re-map the buttons and broadcast the events in the same format you use for the 360 controller. Is that a reasonable approach, you think?

FranticRain, 2016-11-17:
It's impossible to know what will and won't work with this because you are effectively creating middleware for controllers. Each game handles controllers in its own way that can even differ from the same game engine from version to version. The best way to approach this problem would be to implement it and test against games. Some games look explicitly for this driver and only work if the driver is loaded. I know that for a fact because when I was running the Xone-OSX repository, the controllers had identical vendor and product IDs to official 360 controllers, and their button packets were being mangled to be identical to 360 controllers, using a HID Report Descriptor identical to the one used on this driver. Each game is different and has its own quirks. Beyond that, the programming is up to you.

Since this isn't an issue related to the driver itself, and is instead just program structuring advice, I will be closing this issue.

closed 2016-11-17 by FranticRain

SimplGy, 2016-11-17:
That's fair. Middleware is what I'm looking to create. Although it sounds like games are probably querying for HIDs directly, so this middleware is actually looking to hide existing HIDs and re-create HIDs that look exactly like authentic 360 controllers.

Thanks for the advice, Drew and Madd.