360Controller/360Controller #386: Xbox one Controller via Blutooth possible?

jousch, 2016-11-16:
I actually only get the XBOX One Wireless Controller (EU, white, Modelnr. 1708) working with USB cable.

Connecting the controller via macOS Bluetooth prefpanel works, but no controller appears in the 360Controller prefpanel.

Any ideas how to fix? `sudo kextstat` shows that the driver is loaded.

MaddTheSane, 2016-11-16:
The Bluetooth controller does not need the driver when connected via Bluetooth, as it is already handled by macOS' Bluetooth stack. The only thing not handled is force feedback support.

FranticRain, 2016-11-17:
Games must implement support for the Bluetooth controller in their own way. We cannot take a Bluetooth connected controller and make it look like a USB controller. That would be the only way to make these Bluetooth controllers work with existing games. Therefore, there isn't anything we can do. Further discussion about force feedback in regards to the Bluetooth Xbox One controller can be found: #291

closed 2016-11-17 by FranticRain