360Controller/360Controller #389: Xbox one controller not found

nyourchuck, 2016-11-23:
I'm trying to use a controller that shows up in system information, but the system preferences show "no devices found":
  Product ID:	0x561a
  Vendor ID:	0x24c6
  Version:	1.01
  Serial Number:	00005B5BF7B1ED25
  Speed:	Up to 12 Mb/sec
  Manufacturer:	PowerA
  Location ID:	0x14200000 / 1
  Current Available (mA):	500
  Current Required (mA):	Unknown (Device has not been configured)

Let me know if you need other information. Thanks.

FranticRain, 2016-11-23:
This build includes this controller. Please aware of the existing issues with third party Xbox One controllers, especially those from PDP. #333 #325 #312 #300 

https://drive.google.com/file/d/0B5PLCKoMd4zGcDBJSmY3TWdPUG8/view?usp=sharing

closed 2016-11-23 by FranticRain

nyourchuck, 2016-11-25:
Works great now. Thanks.