360Controller/360Controller #393: Randomly stops working

frownedupon, 2016-11-30:
XBox 360 Controller Driver
Version 0.16.4
OSX Sierra 10.12.1
Macbook Pro Retina 13" 2015

The controller works fine but stops working randomly. It stops in the middle of games and when I check in the system preferences settings, the controller is there but the buttons or sticks don't work. The only way to get it back is to unplug it and plug it back in. 

Have tried using different xbox 360 controllers and games.

Cruplezone, 2016-12-03:
loose or broken connection in your controller

justgage, 2016-12-24:
I have two controllers that both do this.

justgage, 2016-12-24:
I think I found the issue with mine. The rubble seems to make it stop working. Not sure why but that's always what makes it happen for me. Here's the logs from `system.log`

```
Dec 23 19:31:38 Gages-MacBook-Pro Docker[542]: VM started at 2016-12-23 19:31:38 -0700 MST
Dec 23 19:31:55 Gages-MacBook-Pro System Preferences[603]: Error: couldn't find ControllerType
Dec 23 19:31:55 Gages-MacBook-Pro System Preferences[603]: No settings found
Dec 23 19:31:56 Gages-MacBook-Pro com.apple.PerformanceAnalysis.animationperfd[251]: BUG in libdispatch client: kevent[EVFILT_MACHPORT] monitored resource vanished before the source cancel handler was invoked
Dec 23 19:32:07 Gages-MacBook-Pro DuetHeuristic-BM[99]: BUG in libdispatch client: kevent[EVFILT_MACHPORT] monitored resource vanished before the source cancel handler was invoked
Dec 23 19:32:14 Gages-MacBook-Pro Screenhero[407]: JS Log: Strophe (1): _dataRecv called
Dec 23 19:32:19 Gages-MacBook-Pro com.apple.xpc.launchd[1] (com.apple.imfoundation.IMRemoteURLConnectionAgent): Unknown key for integer: _DirtyJetsamMemoryLimit
Dec 23 19:32:19 Gages-MacBook-Pro com.apple.xpc.launchd[1] (com.apple.imtranscoding.IMTranscoderAgent): Unknown key for integer: _DirtyJetsamPriority
Dec 23 19:32:19 Gages-MacBook-Pro com.apple.xpc.launchd[1] (com.apple.imtranscoding.IMTranscoderAgent): Unknown key for integer: _DirtyJetsamMemoryLimit
Dec 23 19:32:19 Gages-MacBook-Pro com.apple.xpc.launchd[1] (com.apple.imdmessageservices.IMDMessageServicesAgent): Unknown key for integer: _DirtyJetsamMemoryLimit
Dec 23 19:32:39 Gages-MacBook-Pro System Preferences[603]: No devices found? :( device count==0, i==0
Dec 23 19:32:55 Gages-MacBook-Pro System Preferences[603]: Error: couldn't find ControllerType
Dec 23 19:32:55 Gages-MacBook-Pro System Preferences[603]: No settings found
Dec 23 19:33:05 Gages-MacBook-Pro Screenhero[407]: JS Log: Strophe (1): _dataRecv called
```

FranticRain, 2017-04-17:
This is likely a loose connection. Rumble can easily cause these kinds of disconnects because it can loosen tenuous connections.