360Controller/360Controller #394: Xbox one S controller?

Ext3rminatr, 2016-11-30:
Will this be able to work with the xbox one S controller through bluetooth? I have it downloaded but while trying to connect through bluetooth it always says no devices found

MaddTheSane, 2016-11-30:
The Bluetooth controller doesn’t need this driver, as Microsoft set it up to be a proper Bluetooth device. The only thing missing is force feedback support.

As it doesn’t go through the driver, the preference pane doesn’t recognize it as an Xbox controller, so it isn’t listed there.

This is not true if it’s plugged in via a charge-and-play cable.

imuller, 2016-11-30:
I also have the Xbox One S Controller, and can connect it to my Mac with Bluetooth. But when launch, for example an game in Steam, it doesn't work..

MaddTheSane, 2016-11-30:
Launch Big Picture Mode on Steam and configure the controller.

FranticRain, 2016-12-04:
Games must implement their own support for the Xbox One S controller. Support for controllers in macOS is shoddy at best and most of what we do is a complete shim. This controller works via Bluetooth automatically and works via USB already by using this driver.

closed 2016-12-04 by FranticRain