360Controller/360Controller #395: Unsupported 3rd-party Controller

scosol, 2016-12-01:
Here's the info:

Wired Controller for Xbox 360:

  Product ID:	0x0501
  Vendor ID:	0x0e6f
  Version:	1.00
  Serial Number:	0691C5AA
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Performance Designed Products
  Location ID:	0x14100000 / 9
  Current Available (mA):	1000
  Extra Operating Current (mA):	0

Cheers! :)

FranticRain, 2016-12-02:
This device should already be supported. Check this build just to make sure. If that doesn't work, there may be an issue with your controller. https://drive.google.com/file/d/0B5PLCKoMd4zGcDBJSmY3TWdPUG8/view?usp=sharing

closed 2016-12-02 by FranticRain

scosol, 2016-12-02:
> view?usp=sharing

This build works fine, with two controllers.

I had tried the latest release, along with some of the linked releases in
this issue:

https://github.com/360Controller/360Controller/issues/333

None of them worked- this does- thanks and sweet work! :)

FranticRain, 2016-12-02:
That was an issue related to Xbox One controllers. It has nothing to do with 360 controllers.