360Controller/360Controller #396: Xon-OSX

Oddworld7, 2016-12-03:
Hi, i'm using the old xbox one driver but, it works fine with Mac OS Sierra, the thing is that i have two xbox controllers one i recently use with the xbox and the other one that i haven't played with it for a while, the one i haven't played with connects fine oto my mac and i can play with it, the other one desn't is there a wat to reset the controller from the xbox set up or bluetooth setup just to make it able to use in the mac?

FranticRain, 2016-12-03:
Xone-OSX is deprecated. It is not supported. Do not ask questions about it here. Use 360Controller.

closed 2016-12-03 by FranticRain

Oddworld7, 2016-12-03:
Ok, but i'm not going to use something it half works in Sierra for something already works fine, the question was about the xbox controller after all, but thanks, and thanks for both of the driver.
(Sorry if i sound a bit rude).