360Controller/360Controller #406: Pretend to be an Xbox 360 Controller Option - Not Saving

grpace, 2016-12-23:
I am using an Xbox One S controller via USB and each time I plug it in I have to reselect "Pretend to be an Xbox 360 Controller" from settings. Other than that it works great!

Would love it it if I could get it to detect this controller from Bluetooth.

PizzaPino, 2016-12-26:
same issue

aguynamedryan, 2017-01-06:
First off, thanks for this awesome project!

Same issue that "Pretend to be an Xbox 360 Controller" does not save with a wired POWER A - Mini Controller for Microsoft Xbox One under Mac OS X 10.11.6

FranticRain, 2017-04-17:
This is the same issue as #451 and #280. There was supposed to be a fix for this, but it was never submitted. If someone could look into this further, that'd be fantastic.

FranticRain, 2018-06-01:
Settings bugs should be fixed in the latest release.

closed 2018-06-01 by FranticRain