360Controller/360Controller #408: Sierra and a PowerA xbox one controller

rpwilliamson, 2016-12-27:
trying to get a PowerA xbox one controller working on a OSX Sierra MacBook Air 11" early 2015.

Behavior is that the controller is recognized and the kextstat shows the 360 controller.  The prefs pane shows the "XBox One Wired Controller (Wired)" when the controller is connected, and the light on the controller is on when it is plugged in.  Buttons/triggers/toggles do not trigger anything on the "controller test" tab in the prefs pane, I'm assuming that there should be visible feedback.

Anything I can check?  In theory I can compile the source and step through the code if necessary, but would need guidance on which files/lines should be break-pointed, assuming that I can dredge up my xcode memories.

thanks for any suggestions,
rw

michelleiedelman, 2016-12-31:
I had the same thing happen when I first bought my Microsoft Xbox One Controller and plugged it in to my MacBook Pro 15" (May 2015) with Sierra 10.12.2. 

I realize this isn't a Microsoft controller, but what you're describing matches my experience when my controller's firmware wasn't up to date. 

Is there a firmware update you need to get from somewhere?   

***
For a Microsoft Controller the fix is relatively easy if you have access to Windows or a console -- 

To fix this, I uninstalled the driver in MacOS and then booted into Windows 10 and updated firmware on the controller using Microsoft's Xbox Accessories app (found in App Store).  

If you don't have access to Windows 10, but do have access to Windows 7 or 8, you should be able to download drivers to update the firmware for the controller.  The Xbox Accessories app is only available for Windows 10.  

Another option might be to find someone who has the console and see if plugging it in will get the necessary firmware updates installed. 

After I confirmed the firmware had been updated, I rebooted into macOS, downloaded the driver and everything worked without a problem.  

***

rpwilliamson, 2017-01-01:
Thanks for the suggestion!  He'll try that once he gets back home.

Gozaradio, 2017-01-14:
Can I ask how you got on? I have a wired PowerA Xbox One controller and appear to have the same issue with identical symptoms on OSX El Capitan 10.11.6 (Mid 2011 iMac 27)

I've looked into it a little and PowerA don't appear to offer firmware updates to their controllers.

I don't have immediate access to a Win10 PC or an XBone but I can do if need be.

Thanks

FranticRain, 2017-04-17:
There are known incompatibilities between macOS and both PowerA and PDP devices. I would also like to know if a firmware update fixes these, because it seems like they are very popular, but don't work with the driver for whatever reason.

closed 2018-04-03 by FranticRain

nrobinson2000, 2018-06-28:
It works for me.