360Controller/360Controller #410: Wired Xbox 360 Controller ot Recognized

dahadaller, 2016-12-27:
Product ID:	0x0213
  Vendor ID:	0x0e6f

  So my wired afterglow controller for xbox 360 worked in the driver pane yesterday, but today it's not being recognized at all. I tried reinstalling and uninstalling the driver, but that doesn't seem to work. 

The weird thing is, if I restart the computer (with the driver already installed), the controller will work (in the preferences pane) for a couple of minutes after my mac has started up, but then it just stops working after a period of time.

Vasir1, 2017-02-16:
Have you had any luck? My device has the same product ID and vendor ID as you. Bump?

dahadaller, 2017-02-17:
no I haven't had any luck :( still won't work

FranticRain, 2017-04-17:
This device has already been added to the device list. If it does not currently work with the latest release, then this is another example of the issues PDP controllers have with macOS. We do not see these controllers from the driver, therefore we can do nothing to resolve issues with these controllers. It is unfortunately a bug with macOS and there is nothing we can do to solve it. Sorry.

closed 2017-04-17 by FranticRain

dahadaller, 2017-04-18:
Ok thank's for the heads up :)