360Controller/360Controller #411: PDP Legendary Collection: Deliverer of Truth not found

kittburglar, 2016-12-28:
Here's the information for the PDP Xbox one controller:

Legendary Collection: Deliverer of Truth:
  Product ID:	0x0163
  Vendor ID:	0x0e6f
  Version:	1.02
  Serial Number:	00000239AFA07717
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Performance Designed Products
  Location ID:	0x06400000 / 3
  Current Available (mA):	500
  Extra Operating Current (mA):	0

AdamSkuse, 2017-01-10:
Having same issue with a PDP controller:
 Product ID:	0x0501
  Vendor ID:	0x0e6f
  Version:	1.00
  Serial Number:	06922FBD
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Performance Designed Products
  Location ID:	0x14100000 / 24
  Current Available (mA):	1000
  Extra Operating Current (mA):	0

El Capitan 10.11.06 on a Macbook Pro

FranticRain, 2017-04-17:
For the first device, it has been added. It should be in a new release that will come out in a few days. As for the second device, it has already been added to the device list. If it does not currently work with the latest release, then this is another example of the issues PDP controllers have with macOS. We do not see these controllers from the driver, therefore we can do nothing to resolve issues with these controllers. It is unfortunately a bug with macOS and there is nothing we can do to solve it. Sorry.

closed 2017-04-17 by FranticRain