360Controller/360Controller #412: Similar issuer with wired Xbox One controller on Mac.

DeadZoker, 2017-01-02:
I just bought a PDP Xbox One controller. I had follow every basic instruction when first install the Drive but nothing happen. I had my applicate in system preference but it just unable to find the connect control.The controller doesn't even light up , it give me "no device found" in the test controller. I read the "Xbox Controller Driver for macOS" that you upload and check the "Wired Xbox One Controllers" section. I did go to my system report to see does it connect to my laptop or not and  it is and yet for some reason it doesn't even had any respond. So what do you think is wrong with the controller?

DeadZoker, 2017-01-02:
I am using a data cable not a charge cable.I had just check it myself using 2 different cable I had. The data one will show the controller on system report and charge cable won't. So for some reason the controller just doesn't work. It won't light up just like it doesn't even connect to the laptop.

DeadZoker, 2017-01-02:
http://imgur.com/a/XpbQv

JurianBruinink, 2017-01-17:
Same problem, same controller. 10.10.5

FranticRain, 2017-04-17:
Unfortunately, PDP controllers have issues with newer versions of macOS. This device should match with our controller, but macOS refuses to comply. We give macOS a suggested list of devices that the driver should match to, but in the case of devices like this, there is some issue in compatibility between macOS and the hardware that makes it so that the device does not match to our driver. There is nothing we can do about this, sorry.

closed 2017-04-17 by FranticRain