360Controller/360Controller #413: @Play 360 wired contoller not recognized by driver

danieldh206, 2017-01-03:
Looked through the Info.plist and after converting the hex numbers I did not see the product ID but did see the same Vender ID under the game stop controllers. Preference pane is saying no devices found. I am still on OSX 10.10.5
Thanks
 Product ID:	0x0501
  Vendor ID:	0x0e6f
  Version:	1.00
  Serial Number:	06C95AB6
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Performance Designed Products
  Location ID:	0x1d200000 / 2
  Current Available (mA):	500
  Current Required (mA):	Unknown (Device has not been configured)

macaddict89, 2017-01-12:
I have the same issue

FranticRain, 2017-04-17:
Unfortunately, it looks like macOS has some sort of issue with this controller. This product should already match with the driver, but it appears that macOS does not want to pair that device with our driver. If the operating system doesn't even show the driver that the controller exists, we can't do anything about it, sorry.

closed 2017-04-17 by FranticRain