360Controller/360Controller #416: Mac has crashed repeatedly in first day since installing driver

danielfox, 2017-01-07:
Sorry if this is too similar to the 'Kernel panic' thread started by christ776, and should be merged. My macbook air has crashed 5 or 6 times since installing this Xbox controller driver yesterday. It does it periodically, not always while playing a game, or even while the controllers are connected. Here is the latest panic report:

Anonymous UUID:       9E498F98-07AD-8540-748A-17767FB8F4DD

Sat Jan  7 02:48:13 2017

*** Panic Report ***
panic(cpu 2 caller 0xffffff8009205ead): Kernel trap at 0xffffff800969e387, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x00000000000005a8, CR3: 0x000000000c873000, CR4: 0x00000000003627e0
RAX: 0x0000000000000000, RBX: 0xffffff8018d17c00, RCX: 0xffffff8018cc65c0, RDX: 0xffffff8014513040
RSP: 0xffffff90a46c3e10, RBP: 0xffffff90a46c3e10, RSI: 0x0000000000000000, RDI: 0xffffff8018d17c00
R8:  0x000000000000ffff, R9:  0x0000000000010001, R10: 0xffffff80098671b0, R11: 0xffffff800973b664
R12: 0xffffff8018cc6608, R13: 0xffffff8018cc65c0, R14: 0xffffff8014513040, R15: 0xffffff8018cc65c0
RFL: 0x0000000000010246, RIP: 0xffffff800969e387, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x00000000000005a8, Error code: 0x0000000000000000, Fault CPU: 0x2, PL: 0, VF: 1

Backtrace (CPU 2), Frame : Return Address
0xffffff90a46c3aa0 : 0xffffff80090f211c 
0xffffff90a46c3b20 : 0xffffff8009205ead 
0xffffff90a46c3d00 : 0xffffff80090a3743 
0xffffff90a46c3d20 : 0xffffff800969e387 
0xffffff90a46c3e10 : 0xffffff7f8990af84 
0xffffff90a46c3e30 : 0xffffff800969744e 
0xffffff90a46c3e80 : 0xffffff800969893a 
0xffffff90a46c3f10 : 0xffffff80096987a6 
0xffffff90a46c3f60 : 0xffffff800969a1be 
0xffffff90a46c3fb0 : 0xffffff80090a0af7 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOHIDFamily(2.0)[8D5ADCF8-7C6B-3B6D-9C9A-E2850B1A0AB6]@0xffffff7f89908000->0xffffff7f89980fff
            dependency: com.apple.driver.AppleFDEKeyStore(28.30)[EA5D0966-E8EA-337A-98EB-195806E8F723]@0xffffff7f898fd000

BSD process name corresponding to current thread: kernel_task

Mac OS version:
16C67

Kernel version:
Darwin Kernel Version 16.3.0: Thu Nov 17 20:23:58 PST 2016; root:xnu-3789.31.2~1/RELEASE_X86_64
Kernel UUID: 8DF81F81-019F-348E-B47E-40A4484B9E21
Kernel slide:     0x0000000008e00000
Kernel text base: 0xffffff8009000000
__HIB  text base: 0xffffff8008f00000
System model name: MacBookAir7,1 (Mac-9F18E312C5C2BF0B)

System uptime in nanoseconds: 4718999523905
last loaded kext at 14281364818: org.virtualbox.kext.VBoxNetAdp	5.0.14 (addr 0xffffff7f8c0c0000, size 28672)
loaded kexts:
org.virtualbox.kext.VBoxNetAdp	5.0.14
org.virtualbox.kext.VBoxNetFlt	5.0.14
org.virtualbox.kext.VBoxUSB	5.0.14
org.virtualbox.kext.VBoxDrv	5.0.14
com.mice.driver.Wireless360Controller	1.0.0d15
com.mice.driver.WirelessGamingReceiver	1.0.0d15
com.apple.filesystems.autofs	3.0
com.apple.driver.AudioAUUC	1.70
com.apple.driver.AGPM	110.23.14
com.apple.driver.ApplePlatformEnabler	2.7.0d0
com.apple.driver.X86PlatformShim	1.0.0
com.apple.driver.pmtelemetry	1
com.apple.driver.AppleOSXWatchdog	1
com.apple.driver.AppleHDAHardwareConfigDriver	278.56
com.apple.iokit.IOUserEthernet	1.0.1
com.apple.iokit.IOBluetoothSerialManager	5.0.2f4
com.apple.Dont_Steal_Mac_OS_X	7.0.0
com.apple.driver.AppleMuxControl	3.13.78
com.apple.driver.AppleHDA	278.56
com.apple.driver.AppleGraphicsDevicePolicy	3.13.78
com.apple.driver.AppleUpstreamUserClient	3.6.4
com.apple.driver.ACPI_SMC_PlatformPlugin	1.0.0
com.apple.driver.AppleLPC	3.1
com.apple.driver.AppleHV	1
com.apple.driver.AppleBacklight	170.9.10
com.apple.iokit.BroadcomBluetoothHostControllerUSBTransport	5.0.2f4
com.apple.driver.AppleSMCLMU	208
com.apple.driver.AppleIntelSlowAdaptiveClocking	4.0.0
com.apple.driver.AppleFIVRDriver	4.1.0
com.apple.driver.AppleThunderboltIP	3.0.8
com.apple.driver.AppleCameraInterface	5.57.0
com.apple.driver.AppleMCCSControl	1.2.15
com.apple.driver.AppleIntelBDWGraphics	10.2.2
com.apple.driver.AppleIntelBDWGraphicsFramebuffer	10.2.2
com.apple.iokit.IOBluetoothUSBDFU	5.0.2f4
com.apple.driver.CoreStorageFsck	540.30.1
com.apple.driver.AppleFileSystemDriver	3.0.1
com.apple.AppleFSCompression.AppleFSCompressionTypeDataless	1.0.0d1
com.apple.AppleFSCompression.AppleFSCompressionTypeZlib	1.0.0
com.apple.BootCache	40
com.apple.filesystems.hfs.kext	366.30.3
com.apple.driver.AppleTopCaseHIDEventDriver	104
com.apple.driver.AirPort.Brcm4360	1130.3.1a4
com.apple.driver.AppleSmartBatteryManager	161.0.0
com.apple.driver.AppleRTC	2.0
com.apple.driver.AppleACPIButtons	5.0
com.apple.driver.AppleHPET	1.8
com.apple.driver.AppleSMBIOS	2.1
com.apple.driver.AppleACPIEC	5.0
com.apple.driver.AppleAPIC	1.7
com.apple.nke.applicationfirewall	172
com.apple.security.quarantine	3
com.apple.security.TMSafetyNet	8
com.apple.iokit.IOUSBUserClient	900.4.1
com.apple.kext.triggers	1.0
com.apple.driver.DspFuncLib	278.56
com.apple.kext.OSvKernDSPLib	525
com.apple.driver.AppleSSE	1.0
com.apple.driver.AppleSMBusPCI	1.0.14d1
com.apple.driver.AppleGraphicsControl	3.13.78
com.apple.driver.IOPlatformPluginLegacy	1.0.0
com.apple.driver.X86PlatformPlugin	1.0.0
com.apple.driver.IOPlatformPluginFamily	6.0.0d8
com.apple.driver.AppleBacklightExpert	1.1.0
com.apple.iokit.IOBluetoothHostControllerUSBTransport	5.0.2f4
com.apple.iokit.IOBluetoothHostControllerTransport	5.0.2f4
com.apple.iokit.IOBluetoothFamily	5.0.2f4
com.apple.iokit.IOSlowAdaptiveClockingFamily	1.0.0
com.apple.driver.AppleThunderboltEDMSink	4.1.1
com.apple.driver.AppleThunderboltDPOutAdapter	4.6.1
com.apple.driver.AppleSMC	3.1.9
com.apple.driver.AppleSMBusController	1.0.14d1
com.apple.driver.AppleHDAController	278.56
com.apple.iokit.IOHDAFamily	278.56
com.apple.iokit.IOAudioFamily	205.11
com.apple.vecLib.kext	1.2.0
com.apple.driver.AppleIntelLpssUARTv0	3.0.60
com.apple.driver.AppleIntelLpssUARTCommon	3.0.60
com.apple.iokit.IOSerialFamily	11
com.apple.iokit.IOSurface	153.3
com.apple.iokit.IONDRVSupport	513.1
com.apple.AppleGraphicsDeviceControl	3.13.78
com.apple.iokit.IOAcceleratorFamily2	289.27
com.apple.iokit.IOGraphicsFamily	513.1
com.apple.iokit.IOSCSIArchitectureModelFamily	394.30.2
com.apple.driver.usb.IOUSBHostHIDDevice	1.1
com.apple.driver.usb.cdc	5.0.0
com.apple.driver.usb.networking	5.0.0
com.apple.driver.usb.AppleUSBHostCompositeDevice	1.1
com.apple.driver.usb.AppleUSBHub	1.1
com.apple.driver.CoreStorage	540.30.1
com.apple.filesystems.hfs.encodings.kext	1
com.apple.driver.AppleMultitouchDriver	368.7
com.apple.driver.AppleInputDeviceSupport	76.4
com.apple.driver.AppleHIDKeyboard	197
com.apple.driver.AppleHSSPIHIDDriver	49
com.apple.driver.AppleThunderboltDPInAdapter	4.6.1
com.apple.driver.AppleThunderboltDPAdapterFamily	4.6.1
com.apple.driver.AppleThunderboltPCIDownAdapter	2.1.1
com.apple.driver.AppleXsanScheme	3
com.apple.iokit.IONVMeFamily	2.1.0
com.apple.driver.AppleThunderboltNHI	4.1.8
com.apple.iokit.IOThunderboltFamily	6.3.3
com.apple.iokit.IO80211Family	1200.12.2
com.apple.driver.mDNSOffloadUserClient	1.0.1b8
com.apple.iokit.IONetworkingFamily	3.2
com.apple.driver.corecapture	1.0.4
com.apple.driver.AppleHSSPISupport	49
com.apple.driver.AppleIntelLpssSpiController	3.0.60
com.apple.driver.AppleIntelLpssI2C	3.0.60
com.apple.driver.AppleIntelLpssDmac	3.0.60
com.apple.driver.AppleIntelLpssGspi	3.0.60
com.apple.driver.usb.AppleUSBXHCIPCI	1.1
com.apple.driver.usb.AppleUSBXHCI	1.1
com.apple.driver.usb.AppleUSBHostPacketFilter	1.0
com.apple.iokit.IOUSBFamily	900.4.1
com.apple.driver.AppleUSBHostMergeProperties	1.1
com.apple.driver.AppleEFINVRAM	2.1
com.apple.driver.AppleEFIRuntime	2.0
com.apple.iokit.IOHIDFamily	2.0.0
com.apple.iokit.IOSMBusFamily	1.1
com.apple.security.sandbox	300.0
com.apple.kext.AppleMatch	1.0.0d1
com.apple.driver.AppleKeyStore	2
com.apple.driver.AppleMobileFileIntegrity	1.0.5
com.apple.driver.AppleCredentialManager	1.0
com.apple.driver.KernelRelayHost	1
com.apple.iokit.IOUSBHostFamily	1.1
com.apple.driver.AppleBusPowerController	1.0
com.apple.driver.DiskImages	444.30.5
com.apple.iokit.IOStorageFamily	2.1
com.apple.iokit.IOReportFamily	31
com.apple.driver.AppleFDEKeyStore	28.30
com.apple.driver.AppleACPIPlatform	5.0
com.apple.iokit.IOPCIFamily	2.9
com.apple.iokit.IOACPIFamily	1.4
com.apple.kec.Libm	1
com.apple.kec.pthread	1
com.apple.kec.corecrypto	1.0
Model: MacBookAir7,1, BootROM MBA71.0166.B13, 2 processors, Intel Core i5, 1.6 GHz, 4 GB, SMC 2.26f2
Graphics: Intel HD Graphics 6000, Intel HD Graphics 6000, Built-In
Memory Module: BANK 0/DIMM0, 2 GB, DDR3, 1600 MHz, 0x80CE, 0x4B3445384533303445452D45474345000000
Memory Module: BANK 1/DIMM0, 2 GB, DDR3, 1600 MHz, 0x80CE, 0x4B3445384533303445452D45474345000000
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x117), Broadcom BCM43xx 1.0 (7.21.171.68.1a4)
Bluetooth: Version 5.0.2f4, 3 services, 17 devices, 1 incoming serial ports
USB Device: USB 3.0 Bus
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Air, Apple Inc., 27.2

danielfox, 2017-01-16:
Actually, I was using wireless controllers, and I think you can avoid these crashes by turning off the controllers (e.g. in System Preferences) before disconnecting the wireless adaptor. I haven't been experiencing the crashes lately, and I'm pretty sure that's what I've been doing differently.

justinbmiller, 2017-02-28:
For what it's worth, this worked for me as well. I deleted the Android File Transfer app as per an older GitHub thread, which installed with my Wacom tablet. This did not fix the issue. I made sure I was running this latest version and had the latest Sierra update (10.12.3). 

When I powered off the wireless 360 controller and then closed the lid of my MBP (both after removing the USB device and leaving it in) the machine did not experience the kernel panic and crash. Thanks for posting your additional follow up. I was ready to completely remove the program and give up.

nagybal, 2017-06-28:
Issue still exist for too long now. It would be might better to roll back the driver version to beta.