360Controller/360Controller #417: Buttons are "sticking" in the system

mandersondesign, 2017-01-12:
When I press buttons on the controller, the recognition of the input "sticks" so that it seems like the x or a buttons just are being held down, when i am clicking them briefly.  See the below GIF to see the behavior.

http://www.giphy.com/gifs/l0JM7pCOVcH5LCkIE

Using a newer xbox controller (with 3.5mm port) and on macOS Sierra 10.12.2

This issue seems to be only happening now that I upgraded to Sierra from El Capitan.

Anyone seen this and fixed it?

sum41foxdie, 2017-11-22:
Same issue in High Sierra, maybe downgrade the OS is the only way for now.

FranticRain, 2018-06-01:
I spent several minutes trying to recreate this issue by lightly tapping the "A" button on my Xbox One controller and I couldn't ever get it to replicate the issue. Is this fixed in the latest release of the driver, or are you still experiencing it?

FranticRain, 2018-12-04:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-12-04 by FranticRain