360Controller/360Controller #421: Remap stick and trigger axis

xythobuz, 2017-01-20:
I'm still trying to get this controller diver to properly work with the GTA:SA version available on Steam for Mac. It basically seems to work, but the axis mapping is wrong: the right-stick X and Y axes are swapped with the LT and RT axis, respectively.

I saw swapping the two sticks is already implemented, so I think it should be more or less straight-forward to implement swapping the triggers and a stick.

Thanks!

FranticRain, 2017-04-17:
It actually isn't trivial to swap the two because of how they are implemented in the backend with the HID spec. You'll need to consider a user-space remapping tool to resolve this issue.

closed 2017-04-17 by FranticRain

FranticRain, 2017-04-17:
This may possibly be resolved if someone implements the proposal in #455