360Controller/360Controller #427: Controller is not working

hsabrey, 2017-02-07:
after downloading the driver from here v0.16.04 and installing is successfully, when injecting the USB wireless controller the led is not blinking, system setup: 

- MacBook Air Late 2011
- Xbox 360 controller which is shipped withe oculus rift.
- MacOS Sierra v10.12.3

how to more take is the USB port is getting signal for the dongle or not?

FranticRain, 2017-04-17:
Didn't the Oculus Rift ship with an Xbox One controller? This driver does not support the Xbox One wireless dongle. The protocol is very complex and no one has cracked it yet. We're sorry for the inconvenience, but if you want to use an Xbox brand controller wirelessly, we currently recommend using a Bluetooth Xbox One controller, which works directly with macOS and does not require this driver.

FranticRain, 2018-04-03:
Closing due to inactivity.

closed 2018-04-03 by FranticRain