360Controller/360Controller #433: Xbox 360 Wireless Controller Searching for Connection Despite Being Connected

lebleu29, 2017-02-17:
Weird. So, it works great for me except even though it's connected, the light on the xbox button continues to flash in a circle which normally indicates it's looking for a connection, right? It stops and connects briefly and then continues flashing. Problem is that the controller turns off automatically while I'm playing a game. Running Sierra.

FranticRain, 2017-04-17:
This is a semi-known bug with the wireless implementation. We have to send a packet to tell it to change the LED pattern, but for some reason it doesn't like to work using Sierra. I'll look into it if I have time, but I really know very little about the wireless implementation of this driver.

Dniel97, 2018-04-24:
Did you manage to solve this "bug"?

FranticRain, 2018-04-24:
No, but if your controller has batteries in it, removing the batteries should resolve this issue.