360Controller/360Controller #434: Controller Connects for a second, then disconnects right away

jeffcandoris, 2017-02-18:
my xbox 360 wireless controller connects for a second and then disconnects right away. it registers that I am pressing buttons in that 1 second it is connected, but loses connection. I'm using a ZettaGuard receiver. Help? Thanks.

wiliamding, 2017-03-01:
I found this issue may cause of the USB power manage of the OSX system, It works after I pull out all of my other USB devices(a mechanical keyboard，a hub,a external hard driver).

ljrzarate, 2017-03-05:
I have a similar issue, I do not have any other device connected to my laptop and the controller stop working, the dropdown show the controller but none of the buttons works. it happen each time I connect the controller.
Hope someone help soon :)