360Controller/360Controller #438: Xbox 1 controller not working

AlexFox12787, 2017-02-26:
I'm using a xbox one S controller when I plug it in it is not detected by the driver, if i connect it via bluetooth steam recognizes it and i can play games on it, but it doesn't work wired.
I'm running a macbook pro with osx sierra

pepitito0, 2017-03-12:
Do you have the game battleblocktheater because my Xbox one s controller works with steam big picture but I haven't tested the controller with any other game

FranticRain, 2017-04-17:
Please provide your controller's vendor and product ID's as described in the README. This may be a slightly newer controller, because this controller should be recognized just fine by the driver. If the VID and PID for the controller cannot be found in macOS, try a different micro USB cable, just in case.

FranticRain, 2018-04-03:
Closing due to inactivity.

closed 2018-04-03 by FranticRain