360Controller/360Controller #447: trigger axis not working as designed

Deadeye-X7, 2017-03-23:
Hi there,

First of all, totally thrilled that this exists. A sincere thanks for your hard work.

I'm a retro gamer and managed to get the latest driver working with a third party xbox wireless receiver and legit xbox 360 controller. I have Descent 2 working on my Mac which works great with the controller for the most part. only trouble is that when assigning the trigger axis to move me up or down (left trigger, or right trigger) it only identifies the one trigger I pull to make the assignment in game, then calls that single trigger the z axis + and -, instead of making one trigger z+ and the other z- as it works in most games I've played. 

In Descent 2, it causes the ship to sink rapidly when the trigger is let go. To stay level, you have to hold it 50% of the way down, setting the 'z axis' to even.

Thought I would draw some attention to this since it doesn't seem to be correct behavior. I haven't tested too many other games that need a z axis this way, so I'm not sure if this is a wider issue.

If there's a way to correct this that I haven't discovered yet, please let me know.

Once again, appreciate the effort here. Thanks so much.

MacDragon, 2017-03-23:
This is the correct behaviour for the age of game, back when it was designed controllers didn't seperate triggers onto seperate inputs and they were treated as one under dinput, nowaadays with xinput, they do, and as this controller driver is exposing xinput, each trigger is a seperate independent axis.

FranticRain, 2017-04-17:
Yeah, sorry. This isn't something the driver can fix. its a game level issue. Sorry.

closed 2017-04-17 by FranticRain