360Controller/360Controller #449: Xbox One S controller

Fredje2001, 2017-04-02:
I woud like if there was eddit suport for bluetooth conected controlers we now we need to wire the controler to the mac

Cruplezone, 2017-04-08:
The argument is that it works natively. Phaha in which games I have no idea other than emulators as most 'controller-supported games' for mac specifically look for this driver.

FranticRain, 2017-04-17:
This is correct. This is a USB driver, not a Bluetooth driver. We cannot spoof a Bluetooth controller to look like a USB one, and as such there is nothing we can do. We realize that this is inconvenient for back-compatibility, but this is a game-level issue. Theoretically if games support the DS4, and they should also support the Bluetooth Xbox One controllers.

closed 2017-04-17 by FranticRain