360Controller/360Controller #451: [BUG?] does not remember states

lewislepton, 2017-04-09:
this is something ive noticed. mainly because ive been doing some digging on why some games cant be played with a controller, even though it says they can.

i have an xbox one controller, but that seems to be a problem with some games, mostly gamemaker based. but have ran into others.
but i remembered that this driver has the setting to make the XO controller act like a 360 controller. that did help. BUT, there is a problem with the driver in that it does not seem to save the states in which you choose the 360 option

here are 3 screenshots. first showing my controller not plugged in, then is plugged in with the 360 option clicked & lastly when the system preferences are closed & straight away reopened [with controller still plugged in], shows that the 360 state is not saved.
on top of that it does not remember deadzone states [though i did not set them in the pictures] as well when they are set & when the system preferences are closed. you have to leave it open to keep the effect of 'tricking' the xbox one controller

[image]
[image]
[image]

lewislepton, 2017-04-09:
this is just using the released version. not built from source. though may just try that now to see

FranticRain, 2017-04-17:
This was supposed to be solved with #280, but unfortunately no work was ever completed. I'll look into it if I have time.

lewislepton, 2017-04-17:
cool. cheers man.
been doing some digging for people regarding why do things not work etc & this seems to be 1 such thing. its ok if kept open.
but then how some people code a game can be shocking ;)

Jip-Hop, 2017-07-08:
I noticed my dead zone settings aren't remembered either.

thumbtax, 2017-09-20:
I haven't seen this spelled out (though larianstudios essentially details this in #280), so leaving this here.
I was experiencing the same issue with settings not being saved.

When I looked in /Library/Preferences, I couldn't find the com.mice.driver.Xbox360Controller.devices.plist file referenced in #280.

**Work-around**
I was able to get around the issue by:

- creating a new _empty_ .plist file
- calling it 'com.mice.driver.Xbox360Controller.devices.plist'
- sticking it in /Library/Preferences
- opening SysPref -> xbox 360 controllers
- and closing the window

This populated the empty .plist file with all of the current settings.
After that, setting changes were saving properly.

As larianstudios mentioned in #280, this appears to be a permission issue.
In my case, not enough permission to _create_ the .plist file.
Looks like the daemon can read/write just fine though.

RodrigoCard, 2017-11-01:
So @thumbtax if this is just a premission issue, then adding this line:

> `/usr/bin/touch /Library/Preferences/com.mice.driver.Xbox360Controller.devices.plist`

to the `./Install360Controller/Scripts/finish.sh` file should fix the issue for new installations, in theory.
I cant test it right now because I dont have a proper Xcode 6.4 installation (F.U. Apple for breaking that kind of stuff) working in my current environment, but if someone want to try, that would help.

FranticRain, 2018-04-02:
@RodrigoCard Heads up, a fix was found to use the Xcode 6.4 build tools in more recent OSes. https://github.com/360Controller/360Controller/issues/605#issuecomment-372765092

FranticRain, 2018-06-01:
Settings bugs should be fixed in the latest release.

closed 2018-06-01 by FranticRain