360Controller/360Controller #453: My xbox one wired controller wont work

hugh321, 2017-04-13:
I have downloaded everything and have set it up on my mac, however once i plug the USB in, it says it has connected but none of the buttons react with the computer, the small picture model doesn't show any buttons being pressed. How can i fix this so that it will work?

snapopotamos, 2017-04-18:
I have the same problem. I'm using [this controller](https://www.powera.com/products/wired-controller-xbox-one-green/) and OSx Sierra 10.12.3

FranticRain, 2017-04-18:
Have you attempted to update the firmware for you controller? It seems like this issue is caused by outdated controller firmware.

TommyMacMoo, 2017-04-18:
For me, its working perfectly, on a MacPro 6.1 ...Thank you so much for doing this fine piece of Software!!! No problems detected @ the moment, using it since 14 days now, keep it up please, by the way, Im on MacOS Sierra ;)

FranticRain, 2018-05-23:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-05-23 by FranticRain