360Controller/360Controller #454: XBOX One Controller Not Connecting to Mac

louiefuerte1, 2017-04-13:
I keep having this problem and i see others are too but my Xbox One controller will not connect to my MacBook Pro even with a USB wire that i have to plug in.

Cruplezone, 2017-04-16:
Add it yourself using the readme guide. Closed.

FranticRain, 2017-04-17:
What are the product and vendor ID's of your device? Follow the guide in the README. If you can't find the PID and VID with the controller plugged in, then you need a different micro-usb cable. Beyond that, some third party Xbox One controllers have a known incompatibility with newer versions of macOS. There isn't anything we can do about this, unless you elect to send your device to Apple in order to get this fixed.

FranticRain, 2018-04-03:
Closing due to inactivity.

closed 2018-04-03 by FranticRain