360Controller/360Controller #455: Axis Mapping...

phnord, 2017-04-17:
The way this driver maps axis seems to be causing bugs in a lot of games I've played recently.  I think that, from what I'm seeing, most games expect Axis 0 & 1 to be the left stick, Axis 2 & 3to be the right stick, and then the triggers to be 4 & 5?

But for some reason, my 360Controller is mapping Axis 0 & 1 to the left stick, Axis 2 to the left trigger, Axis 3 & 4 to the right stick, and then axis 5 to the right trigger...

Is this the way the axis are supposed to be mapped?  Is there a bug in my installation? o.O

FranticRain, 2017-04-17:
That's the way the sticks are mapped. Unfortunately, this is just a vestigial trait of the original design and we can't change it now. It would break too many games. Its possible to change this, but not on the fly, so it would require either some really complicated and user-unfriendly steps to dynamically change. Likely it would be a separate install. We've talked about this issue before and decided that its really a game-level issue. Complicating things by creating a second version of the driver only increases headaches for everyone.

phnord, 2017-04-17:
I'm kinda confused... would it be that problematic to have, like... similar to the button remapping feature / the swap sticks feature, an option in the control panel that looks something like:

Axis Mapping:

Left Stick L/R: Axis 0
Left Stick U/D: Axis 1
Left Trigger: Axis 2
Right Stick L/R: Axis 3
Right Stick U/D: Axis 4
Right Trigger: Axis 5

Wherein the Axis are drop-down lists to choose from?

FranticRain, 2017-04-17:
The axis mappings are defined by a HID Report Descriptor. This file is a constant file that is compiled and loaded when the driver is loaded. Its not simplistic to swap between these at runtime. Forcing a change of this would require a setting, then a force reload of the driver. While this is technically possible, it isn't easy, not are there any guaranteed ways to force this to happen. You can theoretically do with this `kextunload`, but calling shell commands from C++ has caused us issues before. 

Basically, here's an implementation that could work. You develop X many HID report descriptors to match each potential combination of sticks and triggers. Then you add this dropdown to the preference pane. The preference pane would set a setting (which sometimes don't actually work on Sierra, that's a bit of a dependency bug) to tell the driver which report descriptor to load. Then it would call `kextunload` on the driver, then `kextutil` to load the driver once again. When the driver loads the report descriptor, it checks the settings and loads the proper report descriptor. This theoretically could work, but I have no interest in developing or maintaining it. We have previously agreed that the esxitingremapping features were a mistake and should not be a part of this driver. If you or someone else would like to implement and maintain this feature, be my guest. I'll reopen this for the time being.