360Controller/360Controller #457: Xbox One Controller does not work on MacOS Sierra

eredhel64, 2017-04-22:
Been through every step of installation and installed the driver to no success.
The ID of the controllers is as follow:

Controller:

  ID do Produto:	0x02d1
  ID do Revendedor:	0x045e  (Microsoft Corporation)
  Versão:	2.03
  Número de Série:	7EED8EC99699
  Velocidade:	Até 12 Mb/seg
  Fabricante:	Microsoft
  ID da Localização:	0x14100000 / 10
  Corrente Elétrica Disponível (mA):	500
  Corrente Elétrica Requerida (mA):	500
  Corrente de Funcionamento Extra (mA):	0

Can anyone help me?
Thanks in advance

FranticRain, 2017-05-07:
This device is definitely supported by the driver. Make sure to upgrade the firmware for this controller. It has resolved issues similar to this one.

FranticRain, 2018-04-03:
Closing due to inactivity.

closed 2018-04-03 by FranticRain