360Controller/360Controller #458: OS X El Capitan. Controller not responding in preference pane menu. Driver Installed okay and Controller Recognized okay.

Chancehadme, 2017-04-27:
Installed Driver Version 0.16.5

LED on my Xbox Controller changed from blinking to solid.  

Opened Preference Pane  (Apple Menu -> System Preferences -> Xbox 360 Controllers). 

Controller  is recognized okay as:

1. Xbox One Wired Controller (Wired)

However, the controller is not responsive. No buttons in the prefs pane light up when I push them, and on the "advanced" tab the left and right stick are also unresponsive. 

Here are my stats:

Mac OS X El Capitan
Version 10.11.6 

Xbox ONE liquid metal controller (Wired/USB):
  Product ID:	0x543a
  Vendor ID:	0x24c6
  Version:	1.01
  Serial Number:	0000D480B38067C8
  Speed:	Up to 12 Mb/sec
  Manufacturer:	PowerA
  Location ID:	0x14310000 / 9
  Current Available (mA):	1000
  Current Required (mA):	500
  Extra Operating Current (mA):	0

FranticRain, 2017-05-07:
This is a known issue with third party Xbox One controllers. Unfortunately, they do not report themselves to the driver properly. This is a bug with macOS and the driver can't do anything about it. If you can, please update the firmware for you controller, as it may resolve this issue.

Chancehadme, 2017-05-26:
Just wanted to follow up for anyone with the same issue. 

Got it working using an xbox 360 wireless controller, and the Microsoft wireless usb adapter (search amazon) to connect it to Mac OS X El Capitan.

closed 2018-04-03 by FranticRain