360Controller/360Controller #463: Wireless controller not working

zmokhtar, 2017-05-10:
Hi, I'm using a "Wireless Adaptor for Windows" but I can't seem to pair controllers or get anything to work.  When I plug in the usb dongle, the light doesn't come on, and pressing the pair button doesn't seem to make a difference.  

Any suggestions?

FranticRain, 2019-05-13:
Wireless controller support has been re-added in the alpha-3 pre-release of this driver. Let me know if this fixes your issue.

FranticRain, 2019-05-26:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-05-26 by FranticRain