360Controller/360Controller #467: Xbox One Controller not working on MacOS X Sierra(10.12.2)

allienable, 2017-05-25:
Hello, I just bought an Xbox One Controller(wired) to play with on my new mac using Sierra OS X(10.12.2). The weird thing is that the driver recognize my controller, but in the Controller Test panel, when I click on something on my controller, nothing happens. Somebody can explain me what to do ? Or is there any way to contact developers so they can fix this bug.

Thanks.

IchSchreiGleich, 2017-07-01:
Same here. The recently bought controller was detected, but nothing happens if buttons where pressed. Connecting an older version of an original microsoft xbox one controller works fine.

FranticRain, 2018-04-03:
Before we can help you, please consult the README and provide all necessary information so that we can help you with your issue. https://github.com/360Controller/360Controller#xbox-one-controllers-connected-with-usb

FranticRain, 2018-05-17:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-05-17 by FranticRain