360Controller/360Controller #468: Xbox one controller not connecting to driver

westfallj19, 2017-05-25:
I checked if my controller was connected to the computer in the USB overview in system report. It said it was connected in there but the driver didn't pick it up. Any help?

Mirageofmage, 2017-07-05:
This is a known issue as seen in #458

FranticRain, 2018-04-03:
Before we can help you, please consult the README and provide all necessary information so that we can help you with your issue. https://github.com/360Controller/360Controller#xbox-one-controllers-connected-with-usb

FranticRain, 2018-05-17:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-05-17 by FranticRain