360Controller/360Controller #469: PDP Camo Wired XBOX One controller registers input sometimes but not others

lynchy901, 2017-05-27:
My wired xbox One controller's input is sometimes picked up by the driver but sometimes it isn't. The controller is always detected by the driver in preferences but most of the time the input doesn't work. I usually just end up unplugging and replugging until, eventually, it starts working. From that point on the controller works great as long as I don't unplug it anymore. I am trying to figure out what I am doing differently to make the input start working.

I have verified the controller works on Windows so it isn't faulty.

My OS is macOS Sierra 10.12

lynchy901, 2017-06-07:
I seem to have found a way to reliably get the controller to register input. In case someone else has this problem try the following steps:

1. Open xbox360controller preferences
2. Plug in controller and it should show up in the dropdown but input does not work
3. Go to 'Advanced' tab
4. Uncheck "Enable Driver" and enter your password when prompted
5. Recheck "Enable Driver" and enter your password when prompted  

For whatever reason this seems to reliably get my controller register input.

nightdr3am, 2017-09-20:
i did that it does not work really

FranticRain, 2018-06-01:
Is this fixed in the latest release? I modified some of the init packets in a way that might fix issues with PDP controllers not initializing properly.

EXPOSEDKING, 2018-10-21:
Yeah pdp controller recognized as wired controller but nothing is responsive in driver or in game

closed 2018-10-21 by FranticRain