360Controller/360Controller #471: 10.12.5 Controller No Longer Detected

JEddie20, 2017-05-29:
Updated to 10.12.5 and I am getting the "no devices found" message in the app. SIP has been disabled which was the fix in 10.12.4 but no longer works with .5

Thanks in advance for the help!

BorisChenCZY, 2017-05-29:
I have the same issue as well :(

FranticRain, 2018-04-03:
What is your controller's PID/VID?

FranticRain, 2018-05-17:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-05-17 by FranticRain