360Controller/360Controller #473: Third party wired controller

t11m77, 2017-05-30:
The following third party wired xb360 controller is unrecognized
`  Product ID:	0xfafa
  Vendor ID:	0x24c6
  Version:	1.00
  Serial Number:	042DDF85
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Performance Designed Products
  Location ID:	0x14200000 / 11
  Current Available (mA):	1000
  Extra Operating Current (mA):	0`

yixuanL, 2017-06-03:
This one is recognised as device but not in the test

PDP Camo Wired Controller for Xbox One：

Product ID：	0x0161
Vendor ID：	0x0e6f
Version：	4.60
Serial Number：	0000077160299853
Speed：	Up to 12 Mb/sec
Manufacturer：	Performance Designed Products
Location ID：	0x14200000 / 4
Current Available (mA)：	500
Extra Operating Current (mA)：	0

Thanks!

dankoB, 2017-06-08:
I have recently purchased a Hyperkin x91 XBox One wired controller. I was under the assumption that this driver supported all XBox One controllers so I never looked into it further before purchasing. I have gone through all possible steps aside from trying to build under unsigned certificates which I would certainly prefer not to do.

[image]

If adding this is at all possible it would be greatly appreciated.

closed 2018-04-03 by FranticRain