360Controller/360Controller #476: macOS Sierra 10.12.5 Controller isn't recognized

kkmini, 2017-06-03:
I went to GameStop today and bought a Xbox 1 Controller for Liftoff on Steam. I tried everything to get it to read the controller, and downloaded 3 different drivers to work with it, nothing worked. Finally I took it back and got a new controller and finally it recognized it with this driver on my mac. However as I'm running Sierra 10.12.5, it doesn't respond, even though it recognizes the device. Super bummed. Is there an update coming?

I do have an older MacBook Air, that is running El Capitan 10.11.6 and it recognized the controller, and it is able to respond in the Device manager. So I know the controller works. Currently downloading steam to that old machine and hoping it will work with Liftoff. Would love to use it on my better and newer mac. Let me know if there is anything else I could install to get it to run.
-K

Mirageofmage, 2017-07-05:
As of (July 5th, 2017) there isn't a fix (#458)

FranticRain, 2018-04-03:
Before we can help you, please consult the README and provide all necessary information so that we can help you with your issue. https://github.com/360Controller/360Controller#xbox-one-controllers-connected-with-usb

FranticRain, 2018-05-17:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-05-17 by FranticRain