360Controller/360Controller #477: XBox One wired controller does not register input on the driver device screen

adsherman, 2017-06-06:
I installed the most recent version (16.5), and my Macbook detects the controller but it will not respond to input and show on the controller driver view panel. I also have a 360 controller and it does seem to be working in the manner that the One controller should be -- am I missing something that I need? I just installed the driver with nothing else

Mirageofmage, 2017-07-05:
This is a known issue, with no current fix (as of July 5th, 2017) #458

FranticRain, 2018-04-03:
Please provide your PID and VID as mentioned in that second of the README.

FranticRain, 2018-05-17:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-05-17 by FranticRain