360Controller/360Controller #486: Xbox One controller not working even after latest firmware update

finndavies, 2017-06-25:
I have had this driver working before flawlessly for quite a few months. I know my cable is a data cable. I know my games are supported. However, after plugging it in this morning I cannot seem to get it detected.

The connection is good, I can tell because Steam is recognising the controller. The driver must be the issue?!? I've tried all typical troubleshooting: Reinstalling the latest version of the driver, changing the cable, even changing the controller. I have also updated my controllers to 2.3.2381.0 using my Xbox. I can't think of anything else to try. 

I hate to add another issue to the long list seemingly the same as everybody else's but I'm pretty sure I've tried everything there is to try?

The driver worked so great before I'm eager to get it working again.

Running OSX El Capitan 10.11.6

Thanks.

Mirageofmage, 2017-07-05:
Have you tried enabling the 360 controller emulation option? 
[image]

Mirageofmage, 2017-07-05:
It seems like this is a known issue as seen in #458

FranticRain, 2018-04-03:
Before we can help you, please consult the README and provide all necessary information so that we can help you with your issue. https://github.com/360Controller/360Controller#xbox-one-controllers-connected-with-usb

Likely this is a PDP/PowerA controller, correct?

FranticRain, 2018-05-17:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-05-17 by FranticRain