360Controller/360Controller #488: Controller recognized, but doesn't work in game!

jaymusic1, 2017-06-27:
Hey, guys!
I'm a newbie in this kind of stuff and I'm having a hard time making my 360 controller to work.
I've installed the driver, the controller works on the Controller Test and Binding, but I'm trying to make it work in Skyrim and have no clue of what to do. The option "Xbox 360 Controller" is allowed in the Gameplay Settings, but nothing happens. I can't change the controls or anything.

Can anyone give me a light?
Thanks :)

Mirageofmage, 2017-07-05:
Are you using a Cider or Wine wrapper to run the game?

oli1220, 2017-08-05:
I have the same issue I am trying to get it to work in wine and steam games but neither of them work they can detect the controllers in the game and the driver but does not respond to any of the buttons that are pressed in either

randyu1, 2017-08-08:
i also have the same issue i connect it to my mac and the light on the controller doesn't turn on. it isn't responsive when its connected to the computer.

yvonnie, 2017-11-25:
I'm having this same issue, running Steam using Crossover. Controller is recognized in system prefs, but not in game.

xbox one & windows wired controller
Mac 10.12.6
Crossover 16.2
Steam (windows) up to date (cant find version #)

FranticRain, 2018-04-03:
The driver does not provide any sort of compatibility for individual games beyond what is provided in the preference pane. The driver simply sends button data to the operating system. Once it does that, control is out of the driver's hands. Since Apple does not provide any standardized means of interacting with controllers in macOS, every game uses a different method to read controller data. We cannot provide individual fixes for each game within the controller. That simply isn't feasible. It is the responsibility of game developers to support controllers, not the other way around. We cannot and will not provide support for individual games. If you are using an Xbox One controller, we can recommend that you use the "Pretend to be an Xbox 360 Controller" option in the "Advanced" menu for better game compatibility. Beyond that, there is nothing we can do.

closed 2018-04-03 by FranticRain

JAMSART, 2019-02-23:
I just downloaded the software and i'm having the same problem.
The game i'm trying to run is castle crashers, which is a game with full xbox 360 controller support.
Even so, when I start the game the controller wont work with it.
I'm running castle crashers through steam, and steam fullscreen works with my controller, but the games do not.