360Controller/360Controller #490: Controller Test not working OS X Yoemite

mayamcgill, 2017-06-28:
I have the latest version and my Xbox one controller lights up when its plugged in. The driver also shows that its plugged in but the controller test doesn't register that any buttons are being pressed.

Mirageofmage, 2017-07-05:
Does the controller still work when being used in games?

oli1220, 2017-08-12:
Same Issue here shows that it connects lights up but doesn't work in game and the driver.

mayamcgill, 2017-09-23:
@Mirageofmage no, it doesn't.

FranticRain, 2018-04-03:
Before we can help you, please consult the README and provide all necessary information so that we can help you with your issue. https://github.com/360Controller/360Controller#xbox-one-controllers-connected-with-usb

FranticRain, 2018-05-17:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2018-05-17 by FranticRain